OpenTabletDriver/OpenTabletDriver #2341: Windows 11 cursor issue

flameboi900, 2022-07-07:
Hello, I recently installed windows 11 and noticed when I resintalled my tablet driver to the latest version of opentabletdriver and put in my area and filters and tried playing like I did in windows 10, my cursor felt extremely laggy like maybe 15-20 fps kind of feeling but this issue was only when opentabletdriver was minimized. This issue has forced me to go back to windows 10 unfortunately and I really wanted to try windows 11 and stay on it but, this issue has now prevented that. If you could either recreate the issue to see it for yourself and or know the problem already and are working on a solution that would be great 👍

jamesbt365, 2022-07-07:
#1999

jamesbt365, 2022-07-07:
also please don't use blank issues with no reason too thank you, use the templates.

flameboi900, 2022-07-07:
my apologies first time giving feedback on a GitHub didn't know thank you