OpenTabletDriver/OpenTabletDriver #2343: Tablet Disconnecting Randomly

RaphiOsu, 2022-07-09:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Steps to Reproduce

Tablet will be working perfectly fine then will randomly disconnect
Tablet stays disconnected with with white and blue lights still signalling the pen is connected
Moving the tablet pen off the area shows the pen has been moved away with the blue light indicator on the tablet but cursor is unresponsive during these movements and indications
Only started experiencing these issues after updating to the new version of OTD and also switching the filter used to remove chatter

### Expected Behavior

Expect tablet pen and cursor to be responsive as the indicators on tablet are active and tablet remains connected in OTD

### Observed Behavior

No cursor response as if tablet is disconnected

### Tablet Device

Wacom CTH-680

### Diagnostic Information

[tablet disconnect.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/9077522/tablet.disconnect.txt)

jamesbt365, 2022-07-09:
"Message": "Unsupported interval detected, will use fallback timer. Expect high CPU usage",

likely caused by using an unsupported interval due to the way windows timers work over stressing the cpu causing drops of input

RaphiOsu, 2022-07-10:
> 

any way to fix this?

gonX, 2022-07-10:
> any way to fix this?

Ensure all your filters are using a timer frequency equal to 1000, 500, 250, 200 or 125

RaphiOsu, 2022-07-10:
> Ensure all your filters are using a timer frequency equal to 1000, 500, 250, 200 or 125

fixed thank you

closed 2022-07-10 by RaphiOsu