OpenTabletDriver/OpenTabletDriver #2344: Support Request for Wacom cintiq 16

nukeCrayon, 2022-07-10:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### What do you need help with?

OpenTabletDriver doesn't detect wacom cintiq DTK1660K0A through wayland.

### What have you tried?

I tried restarting it but it says it doesn't detect cintiq

### Diagnostic Information

[Untitled document.docx](https://github.com/OpenTabletDriver/OpenTabletDriver/files/9078415/Untitled.document.docx)

jamesbt365, 2022-07-10:
upload the file as a txt please.

nukeCrayon, 2022-07-10:
[test.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/9080046/test.txt)
here.

InfinityGhost, 2022-08-26:
Bumping, diag was provided but there was never a response.

jamesbt365, 2022-09-03:
#2384

jamesbt365, 2022-09-03:
Also rmmod wacom

closed 2022-09-03 by jamesbt365