OpenTabletDriver/OpenTabletDriver #2346: Adding an anti-smoothing filter from Hawku

Artikal-afk, 2022-07-13:
### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Description

Not so long ago, the Hawku driver stopped working for me and I switched to OpenTablet, but I was puzzled by the fact that there is no anti-smoothing filter. (Although there are some filters from Hawku, and as far as I know, the Hawku developer participated in the development of OpenTablet) 

This filter helped me quite well when playing Osu and it would be nice to see it here. The filter itself made the cursor much more mobile. With him, there was much less feeling that something was slowing you down, which made it more play pleasant and easier to with him.

jamesbt365, 2022-07-13:
Open the plugin manager, and that is a useless filter anyway. Reconstructor exists.

closed 2022-07-13 by jamesbt365

adryzz, 2022-07-13:
> and as far as I know, the Hawku developer participated in the development of OpenTablet

@Artikal-afk Hawku didn't help writing OpenTabletDriver btw

Artikal-afk, 2022-07-14:
> Откройте менеджер плагинов, и в любом случае это бесполезный фильтр. Реконструктор есть.

Реконструктор крайне кривой в плане работоспособности, особенно на фоне анти-сглаживающего фильтра. Возможно реконструктору нужна доработка, но я не программист. Насчет бесполезности не согласен, мне он явно помогал и могу уверенно сказать, что реконструктор ужасен на фоне анти-сглаживающего фильтра от Хавку.  

The reconstructor is extremely crooked in terms of performance, especially against the background of an anti-smoothing filter. Perhaps the reenactor needs a revision, but I'm not a programmer. I don't agree about the uselessness, he obviously helped me and I can confidently say that the reenactor is terrible against the background of the anti-smoothing filter from Havku.

jamesbt365, 2022-07-15:
>The reconstructor is extremely crooked in terms of performance

Reconstructor is mathematically perfect. You should set it up properly
https://github.com/X9VoiD/VoiDPlugins/wiki/Reconstructor

Whatever translator you are using is probably turning "reconstructor" into "renactor" but it won't really be changed because you it as said above is mathematically perfect.

Hawku "anti smoothing" doesnt really work well at all. Here is a small explanation from the creator of reconstructor.

[image]

This is not planned by anyone to port this filter due to it not exactly being good (this is a fact not an opinion) however feel free to port the filter yourself if you feel like its worth doing so.