OpenTabletDriver/OpenTabletDriver #2353: Latest version of nuget package fails to install

peppy, 2022-07-20:
```
Unable to find package OpenTabletDriver.Native with version (>= 0.6.0.4)
  - Found 2 version(s) in nuget.org [ Nearest version: 0.6.0.2 ]
Package restore failed. Rolling back package changes for 'osu.Framework'.
Unable to find package OpenTabletDriver.Native with version (>= 0.6.0.4)
  - Found 2 version(s) in nuget.org [ Nearest version: 0.6.0.2 ]
```

I'd highly recommend automating the deploy process.

InfinityGhost, 2022-07-20:
@peppy The package is now published.

Keeping this issue open in order to track CI package deployment.

peppy, 2022-07-20:
Thanks. Although it seems I can't update because ns2.1 support is dropped:

```

All packages and projects are compatible with net6.0.
Checking compatibility of packages on .NETStandard,Version=v2.1.
Package OpenTabletDriver 0.6.0.4 is not compatible with netstandard2.1 (.NETStandard,Version=v2.1). Package OpenTabletDriver 0.6.0.4 supports:
  - net5.0 (.NETCoreApp,Version=v5.0)
  - net6.0 (.NETCoreApp,Version=v6.0)
Package restore failed. Rolling back package changes for 'osu.Framework'.
Package OpenTabletDriver 0.6.0.4 is not compatible with netstandard2.1 (.NETStandard,Version=v2.1). Package OpenTabletDriver 0.6.0.4 supports:
  - net5.0 (.NETCoreApp,Version=v5.0)
  - net6.0 (.NETCoreApp,Version=v6.0)
```

InfinityGhost, 2022-07-20:
Oops, somehow I wasn't aware of it being dropped. Will likely push a minor update bringing it back.

InfinityGhost, 2022-07-20:
@peppy Discussion within the Discord has informed me that we've never supported .NET standard. Previous updates were done directly within the csproj, see https://github.com/ppy/osu-framework/blob/master/osu.Framework/osu.Framework.csproj#L50-L55

peppy, 2022-07-21:
I had a feeling something like this was being done somewhere. My bad.

jamesbt365, 2025-01-24:
#3555 and #3550 fixed this for both of our major branches, pushes are now automated on release and our packages are installable!

closed 2025-01-24 by jamesbt365