OpenTabletDriver/OpenTabletDriver #2354: Add support for Huion 580

kaem-e, 2022-07-21:
### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Manufacturer

Huion

### Product Name

580

### Device Width

8" (203.2 mm)

### Device Height

5" (127 mm)

### Pressure Levels

2048

### Lines per Inch (LPI)

4000

### Diagnostic Information

[Untitled.json.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/9155108/Untitled.json.txt)

jamesbt365, 2022-07-21:
Turn off input monitoring and accessibility  for otd, turn them back on export another diag. But because of the way that macos works we'll likely need linux or windows to support it (will work on every platform then)

kaem-e, 2022-07-25:
Sorry for the late response, here i exported another diagnostics report from windows hopefully this'll have the information you need
[asd.json.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/9179766/asd.json.txt)
.

SuspiciousActivity, 2023-04-03:
Any updates on this? I have the Huion 580 as well, I'm on Windows. I have a configuration which recognizes the tablet, but the Tablet Debugger doesn't show any activity.

kaem-e, 2023-04-06:
sadly nope seems like noone has responded so far

SuspiciousActivity, 2023-04-07:
Alright to everyone coming here in the future, I've got it to work by installing the WinUSB driver on interface 1 and 2, not 0, with [this configuration](https://gist.github.com/SuspiciousActivity/9b4ba29475b00dc3ce0b343fe8254c57) and the [release v0.6.0.4](https://github.com/OpenTabletDriver/OpenTabletDriver/releases/tag/v0.6.0.4), working with VMultiMode's relative mode as well. (Put a folder called "Configurations" in the same folder that OTD is in, and put the config in there)

Not creating a pull request since this config can't identify the 580 and would match some random other tablets as well and break things, but as long as you just use the 580, you should be good.

jamesbt365, 2023-04-07:
tablet > device string reader
9580
110
dump all, we can go properly from there.

jamesbt365, 2023-04-07:
Also having winusb on multiple interfaces is problematic, export diagnostic.

kaem-e, 2023-04-07:
Thank you! i'll close this issue then

jamesbt365, 2023-04-07:
That config doesn't work properly and cannot be merged, please follow my steps above.

kaem-e, 2023-04-07:
oh sorry i misunderstood, here

[dump.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/11181017/dump.txt)
[diag.json.txt (i had to add .txt so github would let me upload it, sorry lol)](https://github.com/OpenTabletDriver/OpenTabletDriver/files/11181019/diag.json.txt)

SuspiciousActivity, 2023-04-08:
> That config doesn't work properly and cannot be merged, please follow my steps above.

I know, that's why I didn't make a pull request. The device strings are absolutely useless and I don't want to break anything with my config. It works for me just fine like this, but I know it's a very hacky solution.

kaem-e, 2023-08-21:
@jamesbt365 any updates on this ?

jamesbt365, 2023-08-21:
Requires a pcap with the original drivers to proceed. So follow these instructions.
- Close OTD
- Install the OEM drivers (manufacturer's drivers) for your tablet
- Download and install this tool: USBPcap
- Open C:\Program Files\USBPcap in file explorer (you can copy paste that path in the path bar)
- Start USBPcapCMD.exe
- A terminal window will appear, answer n at the first prompt and reboot your pc
- Open C:\Program  Files\USBPcap
- Start USBPcapCMD.exe
- Screenshot the full contents of the terminal window (you may need to expand the window)

kaem-e, 2023-08-21:
here's the output of USBPcap. 
[image]

jamesbt365, 2023-08-23:
1) press 1 then press enter,

2) enter a name such as capture then press enter,

3) Do the following steps:

Move for 5s
Unplug the tablet from your computer (remove the usb cable from the port),
Wait 2 seconds,
Plug in your tablet again (connect your tablet to your pc ),
Wait 2 seconds,
Move for 5s
Close the console (red cross),
A new file has been generated with the name you gave in the same folder,
Upload that file here

kaem-e, 2023-08-23:
here @jamesbt365, I didn't quite understand what you meant by

> move for 5 seconds

so, I just moved the cursor around for 5 seconds on the tablet first via the pen, removed the cable, and then moved my mouse around for 5 seconds.

(ive had to change the extension to .txt)
[capture.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/12420784/capture.txt)

jamesbt365, 2023-08-23:
Please do it again but without moving the mouse. moving means move the tablet pen.

jamesbt365, 2023-08-23:
@X9VoiD I can't read these, so if you have some time I would appreciate if you took it from here.

kaem-e, 2023-09-15:
@X9VoiD any updates ?

kaem-e, 2023-12-19:
Any updates to this ?

jamesbt365, 2024-01-30:
Then just use the OEM driver and disable high precision mouse in game. 

Unfortunately I do not have time to go through a PCAP to pull out the required information to create a configuration at the moment.