OpenTabletDriver/OpenTabletDriver #2364: Support Request: OTD still does not detect Gaomon 1060.

gianghia123, 2022-07-27:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### What do you need help with?

I have done everything said on #2348 but there were no progress. OTD still does not detect my tablet, despite it show up when I tried lsusb
`Bus 002 Device 002: ID 8087:8000 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:8008 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 004: ID 046d:c31c Logitech, Inc. Keyboard K120
Bus 003 Device 003: ID 1a81:101c Holtek Semiconductor, Inc. wireless dongle
Bus 003 Device 002: ID 256c:0064 Gaomon Gaomon Tablet_1060Pro
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
`

### What have you tried?

I downloaded said file on Discord, made a directory in usr/local, named it OpenTabletDriver. Then in that dir, I made another dir called Configuration, then downloaded the file there, then restarted otd daemon. I also copied and pasted the file's content into a file named '1060 Pro.json' in the dir 'OpenDriverTablet.Configuration/Gaomon' and restarted. Nothing happened, it did not recognize my tablet.

### Diagnostic Information

[diag.json.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/9196612/diag.json.txt)

jamesbt365, 2022-07-27:
Could you get me a string dump then?

Tablet > device string reader
9580
100
empty
dump all
save that file and send here

gianghia123, 2022-07-27:
I cannot do that, it keep displaying this error
[image]
I try to replug, but nothing happened

gianghia123, 2022-07-27:
No use... I tried to send request, and it look like this (with the error)
[image]

jamesbt365, 2022-07-27:
This is the directory bug then. Download the file from latest config branch and move that to `/usr/share/OpenTabletDriver/Configurations`

closed 2022-07-27 by jamesbt365

gianghia123, 2022-07-27:
It does not do anything... I created all the stuff and do all the things, but there is no result
My directory:
[image]

jamesbt365, 2022-07-27:
You pasted the wrong image

gianghia123, 2022-07-27:
Oh sorry, here it is:
[image]

gianghia123, 2022-07-27:
Nope, it is in version 0.6.0.3, the latest, I guess...

jamesbt365, 2022-07-27:
export another diagnostic.

gianghia123, 2022-07-27:
It is over 11pm rn, so I will give you that later. Good night!

gianghia123, 2022-07-28:
Here they are:
[diag3.json.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/9205803/diag3.json.txt)

jamesbt365, 2022-07-28:
I don't see the directory `/usr/share/OpenTabletDriver/Configurations`. Make sure the directory is there if you are running the driver in `/usr/share/OpenTabletDriver`.