OpenTabletDriver/OpenTabletDriver #2365: got suspended after put PC on sleep

Machiron, 2022-07-27:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Steps to Reproduce

After boot up from sleep, I got a message for OTD error that I cant screenshot (sorry) and close it, and the image below is the next error apppeared next:
[image]
and then after close this dialog, OTD self closed, it icon disapear from taskbar.

I did restart OTD after it self close and it work again just fine.

### Expected Behavior

Just yesterday, it'll work just fine after I wake my PC up from sleep. I can play normally on osu! without any issue and I do shut PC down every day.

### Observed Behavior

After boot up, I got error message but I'm not able to catch the first one. I can only provide the second error message one. After it do self close, I did restart OTD and it work again just fine. I'm sure my PC is good and not in lost of electricity or having some bug or issue lately. I think some Windows task did make it suspend itself but I havent have enough knowledge to figure this out. So I put it here just to tell you this bug and you can have some solution for this to make OTD better.

Sorry if I did really miss a report like this one and make a duplicate.

Thank you for reading this, have a good day.

PC spec:
Windows 10 64-bit 21H2 (OS build: 19044.1826, Windows Feature Experience Pack 120.2212.4180.0)
OpenTabletDriver v0.6.0.4 (installed acrodding to the official instruction)

### Tablet Device

VEIKK S640

### Diagnostic Information

[diagnostic.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/9197268/diagnostic.txt)

jamesbt365, 2022-07-27:
Highly likely this is unrelated but i'll leave it open.

gonX, 2022-11-05:
Unless this can be reproduced on `master` (upcoming 0.7 release) I kinda assume this is fixed. Please reopen if it isn't.

closed 2022-11-05 by gonX