OpenTabletDriver/OpenTabletDriver #2367: Local configurations override sometimes doesn't work

jamesbt365, 2022-07-28:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Steps to Reproduce

1) make the folder `$HOME/.local/OpenTabletDriver/Configurations`
2) put a config there
3) sometimes it doesn't work.

### Expected Behavior

It works correctly detecting the directory and using it.

### Observed Behavior

It doesn't do anything, no log, no loading of configs there.

https://discord.com/channels/615607687467761684/723399565780189234/955089890272510053
https://discord.com/channels/615607687467761684/789348845372178482/952317335845367848
a few github issues comments when supporting tablets that show this but i can't be bothered to dig for it.

### Tablet Device

N/A

### Diagnostic Information

N/A

gonX, 2022-11-05:
I'm going to assume this is a user or distro error for now. I've personally never observed this, but have heard about people having issues with it.

closed 2022-11-05 by gonX

gonX, 2025-12-06:
A few years later I think I can safely say that this was caused by daemons being invoked with the `-c` flag.