OpenTabletDriver/OpenTabletDriver #2369: Daemon Error- "Failed to connect to bus: No medium found"

cloudcant, 2022-07-30:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Steps to Reproduce

https://user-images.githubusercontent.com/66269103/181994702-4444a2df-cb1f-4984-bd6b-5ee5961a9562.mp4

I attached a video :)

### Expected Behavior

I have installed opentabletdriver using this method in the past on another arch based distro and everything worked normal, its just expected for that to be the same case on archcraft :/

### Observed Behavior

The daemon no load

### Tablet Device

Veikk s640

### Diagnostic Information

[image]

jamesbt365, 2022-07-30:
`loginctl session-status`

cloudcant, 2022-07-31:
`   ~ loginctl session-status
1 - cloud (1000)
           Since: Sat 2022-07-30 13:08:14 CDT; 1 day 5h ago
          Leader: 535 (sddm-helper)
            Seat: seat0; vc1
         Display: :0
         Service: sddm-autologin; type x11; class user
           State: active
            Unit: session-1.scope
                  ├─    535 /usr/lib/sddm/sddm-helper --socket /tmp/sddm-auth3c6c3d02-a446-4754-8dbe-3062476ce695 --id 1 --start /usr/bin/openbox-session --user cloud --autologin
                  ├─    545 /usr/bin/openbox --startup "/usr/lib/openbox/openbox-autostart OPENBOX"
                  ├─    592 xfsettingsd
                  ├─    595 /usr/lib/xfce-polkit/xfce-polkit
                  ├─    596 xfce4-power-manager
                  ├─    597 picom
                  ├─    628 dunst
                  ├─    630 plank
                  ├─    631 thunar --daemon
                  ├─    643 ksuperkey -e Super_L Alt_L F1
                  ├─    644 ksuperkey -e Super_R Alt_L F1
                  ├─    811 mpd
                  ├─   1770 telegram-desktop --
                  ├─   2259 /usr/bin/flameshot
                  ├─1349176 /opt/discord/Discord
                  ├─1349179 "/opt/discord/Discord --type=zygote --no-zygote-sandbox"
                  ├─1349180 "/opt/discord/Discord --type=zygote"
                  ├─1349182 "/opt/discord/Discord --type=zygote"
                  ├─1349214 "/opt/discord/Discord --type=gpu-process --field-trial-handle=9816158575693437221,12896220476817059625,131072 --disable-features=CookiesWithoutSameSiteMustBeSecure,HardwareMediaKeyHandling,MediaSessionServ>
                  ├─1349220 "/opt/discord/Discord --type=utility --utility-sub-type=network.mojom.NetworkService --field-trial-handle=9816158575693437221,12896220476817059625,131072 --disable-features=CookiesWithoutSameSiteMustBeSecu>
                  ├─1349223 "/opt/discord/Discord --type=broker"
                  ├─1349284 "/opt/discord/Discord --type=utility --utility-sub-type=audio.mojom.AudioService --field-trial-handle=9816158575693437221,12896220476817059625,131072 --disable-features=CookiesWithoutSameSiteMustBeSecure,H>
                  ├─1349340 "/opt/discord/Discord --type=renderer --autoplay-policy=no-user-gesture-required --field-trial-handle=9816158575693437221,12896220476817059625,131072 --disable-features=CookiesWithoutSameSiteMustBeSecure,H>
                  ├─1351029 xfce4-terminal
                  ├─1353726 zsh
                  ├─1353914 alacritty
                  ├─1353921 /bin/zsh
                  ├─1353950 polybar -q main -c /home/cloud/.config/openbox/polybar/default/config.ini
                  ├─1356993 bash /home/cloud/.local/share/Steam/steam.sh
                  ├─1365683 /home/cloud/.local/share/Steam/ubuntu12_32/steam
                  ├─1365690 /home/cloud/.local/share/Steam/ubuntu12_32/steam
                  ├─1365691 ./steamwebhelper --disable-seccomp-filter-sandbox -lang=en_US -cachedir=/home/cloud/.local/share/Steam/config/htmlcache -steampid=1365683 -buildid=1658944613 -steamid=0 -logdir=/home/cloud/.local/share/Ste>
                  ├─1365910 "/home/cloud/.local/share/Steam/ubuntu12_64/steamwebhelper --type=zygote --no-zygote-sandbox --disable-seccomp-filter-sandbox --log-file=/home/cloud/.local/share/Steam/logs/cef_log.txt --product-version=Va>
                  ├─1365911 "/home/cloud/.local/share/Steam/ubuntu12_64/steamwebhelper --type=zygote --disable-seccomp-filter-sandbox --log-file=/home/cloud/.local/share/Steam/logs/cef_log.txt --product-version=Valve Steam Client --l>
                  ├─1365913 "/home/cloud/.local/share/Steam/ubuntu12_64/steamwebhelper --type=zygote --disable-seccomp-filter-sandbox --log-file=/home/cloud/.local/share/Steam/logs/cef_log.txt --product-version=Valve Steam Client --l>
                  ├─1365926 "/home/cloud/.local/share/Steam/ubuntu12_64/steamwebhelper --type=gpu-process --field-trial-handle=1754133597496353603,4396881241635331737,131072 --enable-features=CastMediaRouteProvider --disable-features>
                  ├─1365942 "/home/cloud/.local/share/Steam/ubuntu12_64/steamwebhelper --type=utility --utility-sub-type=network.mojom.NetworkService --field-trial-handle=1754133597496353603,4396881241635331737,131072 --enable-featur>
                  ├─1366013 "/home/cloud/.local/share/Steam/ubuntu12_64/steamwebhelper --type=renderer --disable-seccomp-filter-sandbox --log-file=/home/cloud/.local/share/Steam/logs/cef_log.txt --field-trial-handle=17541335974963536>
                  ├─1366152 "/home/cloud/.local/share/Steam/ubuntu12_64/steamwebhelper --type=renderer --disable-seccomp-filter-sandbox --log-file=/home/cloud/.local/share/Steam/logs/cef_log.txt --field-trial-handle=17541335974963536>
                  ├─1366258 "/home/cloud/.local/share/Steam/ubuntu12_64/steamwebhelper --type=renderer --disable-seccomp-filter-sandbox --log-file=/home/cloud/.local/share/Steam/logs/cef_log.txt --field-trial-handle=17541335974963536>
                  ├─1366263 "/home/cloud/.local/share/Steam/ubuntu12_64/steamwebhelper --type=renderer --disable-seccomp-filter-sandbox --log-file=/home/cloud/.local/share/Steam/logs/cef_log.txt --field-trial-handle=17541335974963536>
                  ├─1383854 zsh
                  ├─1383986 loginctl session-status
                  └─1383987 less

Jul 31 18:41:23 cloud flameshot[2259]: QPainter::setPen: Painter not active
Jul 31 18:41:23 cloud flameshot[2259]: QPainter::setBrush: Painter not active
[... cut ...]
Jul 31 18:41:41 cloud flameshot[2259]: QPainter::setRenderHint: Painter must be active to set rendering hints
Jul 31 18:41:41 cloud flameshot[2259]: QPainter::setCompositionMode: Painter not active
Jul 31 18:41:41 cloud flameshot[2259]: QPainter::translate: Painter not active
Jul 31 18:41:41 cloud flameshot[2259]: QPainter::setPen: Painter not active
Jul 31 18:41:41 cloud flameshot[2259]: QPainter::setBrush: Painter not active
Jul 31 18:41:41 cloud flameshot[2259]: QPainter::setBrush: Painter not active
`

`   ~ echo $DBUS_SESSION_BUS_ADDRESS
unix:path=/run/user/1000/bus
   ~ 

`

cloudcant, 2022-07-31:
sorry about the trash formatting idk how to use github well

jamesbt365, 2022-07-31:
Only thing i can think off is you running `systemctl --user enable opentabletdriver --now` as a user that isn't the same as what the session is running at. Closing as this is an issue with your session or user error and not a driver problem. You can manually start otd with the command `otd` but the issue is with your setup. I would consider looking online.

closed 2022-07-31 by jamesbt365

gonX, 2022-08-01:
Your desktop manager (login manager) is likely broken or improperly configured. Once you get that sorted out, this issue might be relevant: https://github.com/OpenTabletDriver/OpenTabletDriver.Web/issues/43