OpenTabletDriver/OpenTabletDriver #2373: Support Request: Issues while converting wacom tablet area

byczekfernand7603, 2022-07-31:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### What do you need help with?

Hello, after i converted my tablet area from Wacom drivers to OpenTabletDrivers my area feels a little bit smaller than with the previous drivers. This happened to me twice: in july 2021 and july 2022 (now).  
[image]
[image]
I hope my problem is able to be solved.

### What have you tried?

I tried finding help on internet and uninstalling all plugins from OpenTabletDrivers, but it didn't help.

### Diagnostic Information

[nwm.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/9229076/nwm.txt)

jamesbt365, 2022-07-31:
If you used the inbuilt converter by right clicking the area > convert area this is the fact that wacom drivers have inbuilt smoothing so otd feels different because it doesn't.

closed 2022-07-31 by jamesbt365

byczekfernand7603, 2022-07-31:
okay, thank you for your help ^^