OpenTabletDriver/OpenTabletDriver #2377: Fails to load with WACOM Graphire 2 Graphics Art TABLET ET-0405A-U

snovotill, 2022-08-04:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Steps to Reproduce

Connect WACOM Graphire 2 Graphics Art TABLET ET-0405A-U and run OpenTabletDriver.
The driver detects the tablet and then barfs IndexOutOfRangeException:Error as shown below:

[Wacom ET-0405A-U:Info] Output mode: Absolute Mode
[Wacom ET-0405A-U:Info] Display area: [3328x1440@<1664, 720>:0°],
[Wacom ET-0405A-U:Info] Tablet area: [127.6x92.8@<63.8, 46.4>:0°],
[Wacom ET-0405A-U:Info] Clipping: Enabled
[Wacom ET-0405A-U:Info] Ignoring reports outside area: Disabled
[Wacom ET-0405A-U:Info] Tip Binding: [Mouse Button Binding: Left]@0%
[Settings:Info] Driver is enabled.
[IndexOutOfRangeException:Error]        Index was outside the bounds of the array.
   at OpenTabletDriver.Configurations.Parsers.Wacom.Intuos.IntuosTabletReport..ctor(Byte[] report)
   at OpenTabletDriver.Configurations.Parsers.Wacom.Intuos.IntuosReportParser.GetToolReport(Byte[] report)
   at OpenTabletDriver.Configurations.Parsers.Wacom.Intuos.IntuosReportParser.Parse(Byte[] report)
   at OpenTabletDriver.Devices.DeviceReader`1.Main()

### Expected Behavior

Should just load and then the tablet should show in the GUI.

### Observed Behavior

Attempts to load but fails. See steps to reproduce.

### Tablet Device

WACOM Graphire 2 Graphics Art TABLET ET-0405A-U

### Diagnostic Information

{
  "App Version": "OpenTabletDriver v0.6.0.4",
  "Build Date": "5/4/2022",
  "Operating System": {
    "Platform": 2,
    "ServicePack": "",
    "Version": "10.0.22000.0",
    "VersionString": "Microsoft Windows NT 10.0.22000.0"
  },
  "Environment Variables": {
    "USER": null,
    "TEMP": "C:\\Users\\user\\AppData\\Local\\Temp",
    "TMP": "C:\\Users\\user\\AppData\\Local\\Temp",
    "TMPDIR": null,
    "USERPROFILE": "C:\\Users\\user"
  },
  "HID Devices": [
    {
      "DevicePath": "\\\\?\\usb#vid_046d&pid_0893&mi_04#8&e5688e2&0&0004#{dee824ef-729b-4a0e-9c14-b7117d33a817}",
      "Manufacturer": "Unknown Manufacturer",
      "ProductName": "Logitech StreamCam",
      "SerialNumber": "B24BC445",
      "FriendlyName": "Logitech StreamCam",
      "VendorID": 1133,
      "ProductID": 2195,
      "InputReportLength": 0,
      "OutputReportLength": 0,
      "FeatureReportLength": 0,
      "CanOpen": true
    },
    {
[... 1839 lines cut ...]
      "Time": "2022-08-03T22:20:26.5546791-07:00",
      "Group": "DeviceIOException",
      "Message": "Failed to get info.",
      "StackTrace": "   at HidSharp.Platform.Windows.WinHidDevice.RequiresGetInfo(GetInfoFlags flags)\r\n   at HidSharp.Platform.Windows.WinHidDevice.GetSerialNumber()\r\n   at OpenTabletDriver.Devices.HidSharpBackend.HidSharpEndpoint.<>c.<get_SerialNumber>b__19_0(HidDevice d)\r\n   at OpenTabletDriver.Extensions.TryGet[TSource,TValue](TSource source, Func`2 predicate, TValue& value)",
      "Level": 3,
      "Notification": false
    },
    {
      "Time": "2022-08-03T22:20:26.5546856-07:00",
      "Group": "DeviceIOException",
      "Message": "Failed to get info.",
      "StackTrace": "   at HidSharp.Platform.Windows.WinHidDevice.RequiresGetInfo(GetInfoFlags flags)\r\n   at HidSharp.Platform.Windows.WinHidDevice.GetSerialNumber()\r\n   at OpenTabletDriver.Devices.HidSharpBackend.HidSharpEndpoint.<>c.<get_SerialNumber>b__19_0(HidDevice d)\r\n   at OpenTabletDriver.Extensions.TryGet[TSource,TValue](TSource source, Func`2 predicate, TValue& value)",
      "Level": 3,
      "Notification": false
    },
    {
      "Time": "2022-08-03T22:20:26.5546921-07:00",
      "Group": "DeviceIOException",
      "Message": "Failed to get info.",
      "StackTrace": "   at HidSharp.Platform.Windows.WinHidDevice.RequiresGetInfo(GetInfoFlags flags)\r\n   at HidSharp.Platform.Windows.WinHidDevice.GetManufacturer()\r\n   at OpenTabletDriver.Devices.HidSharpBackend.HidSharpEndpoint.<>c.<get_Manufacturer>b__13_0(HidDevice d)\r\n   at OpenTabletDriver.Extensions.TryGet[TSource,TValue](TSource source, Func`2 predicate, TValue& value)",
      "Level": 3,
      "Notification": false
    }
  ]
}

jamesbt365, 2022-09-03:
Could try making a `Configurations` folder next to the executables of otd and moving [this](https://github.com/OpenTabletDriver/OpenTabletDriver/files/9483558/ET-0405A-U.json.txt) here (remove the .txt on the end)

snovotill, 2022-09-06:
Thanks, that helped a little bit in that the tablet is now being enumerated, but it still fails to function apparently due to other array out of bounds errors. Here is the debug log (attached).
[debug.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/9501365/debug.txt)

jamesbt365, 2022-09-06:
The file i sent isn't using the intuos report parser so restart otd.

snovotill, 2022-09-06:
Okay, so I rebooted my PC just be be sure and then I double-clicked on:
C:\ProgramData\OpenTabletDriver\OpenTabletDriver.UX.Wpf.exe

The output is a little different now but still includes an out of bounds error in the tail of the debug log as follows:
```
[Settings:Info]	Presets have been refreshed. Loaded 0 presets.
[Display:Debug]	DPI Awareness enabled
[IndexOutOfRangeException:Error]	Index was outside the bounds of the array.
   at OpenTabletDriver.Configurations.Parsers.Wacom.Intuos.IntuosTabletReport..ctor(Byte[] report)
   at OpenTabletDriver.Configurations.Parsers.Wacom.Intuos.IntuosReportParser.GetToolReport(Byte[] report)
   at OpenTabletDriver.Configurations.Parsers.Wacom.Intuos.IntuosReportParser.Parse(Byte[] report)
   at OpenTabletDriver.Devices.DeviceReader`1.Main()
[IPC:Debug]	Connected to a client.
[Settings:Info]	Driver is enabled.
[Wacom ET-0405A-U:Info]	Tip Binding: [Mouse Button Binding: Left]@0%
[Wacom ET-0405A-U:Info]	Ignoring reports outside area: Disabled
[Wacom ET-0405A-U:Info]	Clipping: Enabled
[Wacom ET-0405A-U:Info]	Tablet area: [127.6x92.8@<63.8, 46.4>:0°],
[Wacom ET-0405A-U:Info]	Display area: [3328x1440@<1664, 720>:0°],
[Wacom ET-0405A-U:Info]	Output mode: Absolute Mode
[Display:Debug]	DPI Awareness enabled
```

snovotill, 2022-09-06:
Farther to above, I wanted to try an older version of the driver but had trouble installing the dependencies in Win11.

X9VoiD, 2023-08-17:
Resolved by #2857

closed 2023-08-17 by X9VoiD