OpenTabletDriver/OpenTabletDriver #2378: Inverted tilt direction on Huion Q11K V2

sajmon170, 2022-08-04:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Steps to Reproduce

1. Open Krita
2. Try drawing with a tilt sensitive brush

### Expected Behavior

The tilt is not inverted in any direction

### Observed Behavior

The Y-axis is inverted when tilting the pen.

### Tablet Device

Huion Q11K V2

### Diagnostic Information

```json
{
  "App Version": "OpenTabletDriver v0.6.0.4+325c28ca",
  "Build Date": "4.08.2022",
  "Operating System": {
    "Platform": 4,
    "ServicePack": "",
    "Version": "5.18.15.1",
    "VersionString": "Unix 5.18.15.1"
  },
  "Environment Variables": {
    "USER": "sajmon",
    "TEMP": null,
    "TMP": null,
    "TMPDIR": null,
    "DISPLAY": ":0",
    "WAYLAND_DISPLAY": "wayland-0",
    "PWD": "/usr/share/OpenTabletDriver",
    "PATH": "/usr/local/bin:/usr/bin:/usr/local/sbin:/home/sajmon/.dotnet/tools:/var/lib/flatpak/exports/bin:/usr/lib/jvm/default/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl"
  },
  "HID Devices": [
    {
      "DevicePath": "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0/0003:256C:006E.0001/hidraw/hidraw0",
      "Manufacturer": "HUION",
      "ProductName": "Huion Tablet",
      "SerialNumber": "",
      "FriendlyName": "Huion Tablet",
      "VendorID": 9580,
      "ProductID": 110,
      "InputReportLength": 12,
      "OutputReportLength": 0,
      "FeatureReportLength": 0,
      "CanOpen": true
    },
    {
      "DevicePath": "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.1/0003:256C:006E.0002/hidraw/hidraw1",
      "Manufacturer": "HUION",
      "ProductName": "Huion Tablet",
      "SerialNumber": "",
      "FriendlyName": "Huion Tablet",
      "VendorID": 9580,
      "ProductID": 110,
      "InputReportLength": 10,
      "OutputReportLength": 0,
[... 1679 lines cut ...]
      "Group": "IPC",
      "Message": "Disconnected from a client.",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2022-08-04T15:59:44.8927485+02:00",
      "Group": "IPC",
      "Message": "Connected to a client.",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2022-08-04T15:59:44.892764+02:00",
      "Group": "Settings",
      "Message": "Presets have been refreshed. Loaded 0 presets.",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    }
  ]
}
```

jamesbt365, 2022-08-04:
Fixed in  #1841

sajmon170, 2022-08-04:
> Fixed in #1841

If it has been fixed in 2021 then what should I do to get it working now? For me, the tilt direction is still inverted. I am running a new version of OpenTabletDriver, so this issue shouldn't be a problem for me.

jamesbt365, 2022-08-04:
The change isn't merged, thus is not fixed in the current version, but the version on that fork.

sajmon170, 2022-08-04:
Should I recompile my drivers using this fork for now? What's stopping this pull request from getting merged?

jamesbt365, 2022-08-04:
>Should I recompile my drivers using this fork for now?

Nothing is stopping you.

>What's stopping this pull request from getting merged?

It could break other tablets, and is better as an attribute instead of changing the parser directly.

q-lum, 2024-01-06:
I'm on a Kamvas 13 and was looking everywhere for a workaround to this issue while the devs are still looking for a satisfying solution, I don't think this info has made it out of the discord yet but the workaround does exist thanks to Kuuuube.

Download and install this plugin: https://github.com/Kuuuube/CustomFilter/releases
then set "Y tilt polynomial" to `-ty`

jamesbt365, 2024-01-06:
Its in the plugin manager now.

wareya, 2025-05-16:
I'm experiencing inverted y tilt on a Huion H950P out of the box on OTD 0.6.5.1. Negating it with the custom filter fixes it for me.

This is what the tablet debugger looks like when I have the tip near the middle and the top of the pen further away from me, forwards and to the right. So the pen is pointed behind me and to my left.

[image]

PR #3701 (2025-02-08): [06x] Invert tilt in UCLogicTiltReportParser
Backport of #1841

Fixes #2378

Affects the following tablets:

Artisul/A1201.json
Artisul/A1201.json
Gaomon/1060 Pro.json
Gaomon/1060 Pro.json
Gaomon/1060 Pro.json
Gaomon/1060 Pro.json
Gaomon/M106K Pro.json
Gaomon/M106K Pro.json
Gaomon/M10K Pro.json
Gaomon/M10K Pro.json
Gaomon/M1220.json
Gaomon/M1230.json
Gaomon/M1230.json
Gaomon/PD1161.json
Huion/H1061P.json
Huion/RTP-700.json
Huion/RTP-700.json
Huion/H1060P.json
Huion/H1060P.json
Huion/H1060P.json
Huion/H1161.json
Huion/H1161.json
Huion/H320M.json
Huion/H320M.json
Huion/H580X.json
Huion/H610 Pro  [... cut]