OpenTabletDriver/OpenTabletDriver #2379: Invisible cursor in certain apps on Wayland with Artist Mode

sajmon170, 2022-08-04:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Steps to Reproduce

Try opening GIMP, Blender, Firefox/Chrome (in XWayland or native Wayland mode) or Photoshop (Wine) under a Wayland session in GNOME. For some reason Krita works without any issues, even though it's not a native wayland program.

I submitted this issue here and not on the GNOME Gitlab page because this problem seems to be dependent on the driver:
1) This problem doesn't exist when using the Absolute Mode
2) After I updated OpenTabletDriver the cursor in Photoshop disappeared. I didn't update anything related to GNOME or its compositor.

### Expected Behavior

The cursor is visible everywhere when using the Artist Mode, just like when using the Absolute Mode.

### Observed Behavior

The cursor disappears in a lot of programs when using a drawing tablet.

### Tablet Device

Huion Q11K V2

### Diagnostic Information

```json
{
  "App Version": "OpenTabletDriver v0.6.0.4+325c28ca",
  "Build Date": "4.08.2022",
  "Operating System": {
    "Platform": 4,
    "ServicePack": "",
    "Version": "5.18.15.1",
    "VersionString": "Unix 5.18.15.1"
  },
  "Environment Variables": {
    "USER": "sajmon",
    "TEMP": null,
    "TMP": null,
    "TMPDIR": null,
    "DISPLAY": ":0",
    "WAYLAND_DISPLAY": "wayland-0",
    "PWD": "/usr/share/OpenTabletDriver",
    "PATH": "/usr/local/bin:/usr/bin:/usr/local/sbin:/home/sajmon/.dotnet/tools:/var/lib/flatpak/exports/bin:/usr/lib/jvm/default/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl"
  },
  "HID Devices": [
    {
      "DevicePath": "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0/0003:256C:006E.0001/hidraw/hidraw0",
      "Manufacturer": "HUION",
      "ProductName": "Huion Tablet",
      "SerialNumber": "",
      "FriendlyName": "Huion Tablet",
      "VendorID": 9580,
      "ProductID": 110,
      "InputReportLength": 12,
      "OutputReportLength": 0,
      "FeatureReportLength": 0,
      "CanOpen": true
    },
    {
      "DevicePath": "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.1/0003:256C:006E.0002/hidraw/hidraw1",
      "Manufacturer": "HUION",
      "ProductName": "Huion Tablet",
      "SerialNumber": "",
      "FriendlyName": "Huion Tablet",
      "VendorID": 9580,
[... 1682 lines cut ...]
      "Group": "IPC",
      "Message": "Disconnected from a client.",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2022-08-04T15:59:44.8927485+02:00",
      "Group": "IPC",
      "Message": "Connected to a client.",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2022-08-04T15:59:44.892764+02:00",
      "Group": "Settings",
      "Message": "Presets have been refreshed. Loaded 0 presets.",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    }
  ]
}
```

jamesbt365, 2022-08-04:
Artist mode is intended to be used without the tip binding bound. Try clearing the binding.

sajmon170, 2022-08-04:
This didn't help - the cursor is still invisible.

jamesbt365, 2022-08-04:
Are you only using one display? Could you show your display setup? Does this only happen when applications are in full screen?

jamesbt365, 2022-08-04:
Similar to #2075 if it doesn't work in fullscreen apps and maybe #1660 related.

gonX, 2022-08-04:
Since it's working in Krita, I don't think this is our bug.

GIMP requires special setup to work correctly for me (regardless of tablet driver and native X11/Xwayland): go to Edit -> Input Devices, and enable Mode: Screen (or Window, depending on your preference) for each relevant device. On Wayland these are  currently the `eraser` and `stylus` xwayland-tablet devices.

I'd suggest double checking with upstream drivers (in-kernel or newest from Digimend) and checking whether the behavior is the same there.

sajmon170, 2022-08-04:
I set up GIMP according to these instructions and while the pen pressure works without any problems, the cursor still disappears when GIMP is maximized. I'll check out the Digimend driver when I have free time but last time I used it (about a month ago) the cursor was invisible in Krita, unlike when using OpenTabletDriver.

gonX, 2022-08-04:
Then I'm guessing this is a GNOME bug actually, since my cursor remains visible when full screened on Sway.

aDelightz, 2023-03-19:
similar issues on sway and hyperland, both fresh installs too although i only tried artist mode on hyprland so far
had no issues before on x11. i think it might have something to do with xwayland? idk
i might do a proper bug report later but tl;dr: most of the time menus worked just fine but then in level/map there'd be weird movement depending on game and how i ran it (sway/hyprland, windowed/fullscreen borderless/fullscreen, absolute/artist mode)

FedFer98123, 2023-09-12:
Just had the same bug on wayland gnome on krita (using krita-wayland fork, not tested on stable krita)