OpenTabletDriver/OpenTabletDriver #2380: Add support for XP-Pen Artist 22E Pro

pjalsGit, 2022-08-06:
### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Manufacturer

XP-Pen

### Product Name

Artist 22E Pro

### Device Width

476.4mm

### Device Height

268.11mm

### Pressure Levels

8192

### Lines per Inch (LPI)

5080

### Additional Information

Note: The device WxH might be incorrect since it was different when I checked on 2 different websites.

### Diagnostic Information

[thing.json.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/9275080/thing.json.txt)

jamesbt365, 2022-08-06:
[Artist 22E.json.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/9275106/Artist.22E.json.txt) move this into `/usr/share/OpenTabletDriver/Configurations` and remove the .txt from the end then `tablet > detect tablet` in otd.

pjalsGit, 2022-08-06:
It detected it but its slightly offset?
[image]
In this image it looks correctly configured though.

jamesbt365, 2022-08-06:
Go into `tablet > tablet debugger` then move your pen around, if you see a blue dot then move your pen to the bottom right of your tablet (ignore the visualizer) and get the highest x and y coordinate value. If you don't see a blue dot then we'll have to change something.

pjalsGit, 2022-08-06:
I saw a blue dot when I moved my pen to the bottom right, but it wasn't in the bottom right in the visualiser, the highest xy is: [47663,26809]

jamesbt365, 2022-08-06:
See if that goes a little higher while i work something out.

pjalsGit, 2022-08-08:
Did you forgor?

jamesbt365, 2022-08-08:
No? I asked if that goes higher, so could you check again.

pjalsGit, 2022-08-08:
Nah, it didn't go higher

jamesbt365, 2022-08-08:
How many pressure levels do you get when pressing your pen all the way down?

jamesbt365, 2022-08-08:
Replace the current configuration with this one and replug your tablet. [Artist 22E.json.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/9282710/Artist.22E.json.txt)

jamesbt365, 2022-08-08:
Once you do that you can check if the pen buttons go to true in the debugger when pressed and if the tablet buttons work if bound.

pjalsGit, 2022-08-08:
The config works perfectly (including buttons), thanks!

pjalsGit, 2022-08-08:
Actually, there might be an issue with tilt, it only goes from `[63,0]` to `[65,0]`. But I don't really use tilt anyway.

jamesbt365, 2022-08-08:
I don't see anything on the product page about tilt, could you get what bytes the tilt is on in the raw data section in the debugger?

pjalsGit, 2022-08-08:
Actually it probably doesn't have tilt, thank you!

jamesbt365, 2022-08-08:
Something should still be done because this could mess with drawing. It should be 0,0 if the tablet has no tilt so it needs to be parsed out,

pjalsGit, 2022-08-08:
Its probably trying to read the last 2 bytes (this is probably not useful at all):
[image]