OpenTabletDriver/OpenTabletDriver #2381: Add support for veikk VK430

kabanery228, 2022-08-07:
### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Manufacturer

veikk

### Product Name

VK430

### Device Width

200

### Device Height

150

### Pressure Levels

8192

### Lines per Inch (LPI)

5080

### Diagnostic Information

[123.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/9277519/123.txt)

jamesbt365, 2022-08-07:
#2227.

closed 2022-08-07 by jamesbt365

pp0236, 2022-09-01:
it said 'No tablets were detected'
but i have vk430 connected

jamesbt365, 2022-09-01:
Because its not in the latest version yet, add the configuration from configs branch with a `Configurations` folder next to the executables of otd.

pp0236, 2022-09-01:
> Because its not in the latest version yet, add the configuration from configs branch with a `Configurations` folder next to the executables of otd.

thanks you, it working now