OpenTabletDriver/OpenTabletDriver #2385: Add support for Woodpad 7

Kirkyeli, 2022-08-12:
### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Manufacturer

ViewSonic

### Product Name

Woodpad 7

### Device Width

200mm

### Device Height

170mm

### Pressure Levels

4096

### Lines per Inch (LPI)

5080

### Diagnostic Information

[a.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/9316135/a.txt)

jamesbt365, 2022-08-12:
Do you have discord, if so could you join [our discord](https://opentabletdriver.net/Discord) as its easier to support tablets that way. If you can't just say and we can continue here.

Kirkyeli, 2022-08-12:
I joined the discord, I may not be responsive though.