OpenTabletDriver/OpenTabletDriver #2387: CTL - 480 Reports with filters not matching.

Panday-Ichihara, 2022-08-15:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Steps to Reproduce

Turn on any filter with hertz frequency, open the game turn on raw input and look at the reports. 

### Expected Behavior

It is expected to report the same amount you put in the filter or little bit less. like 1000hz - 1000reports/s, 500hz - 500 reports/s.

### Observed Behavior

With ctl - 480 reports are less than expected. something like 1000hz - 910 reports/s, 500hz - 420 reports/s.

### Tablet Device

CTL - 480

### Diagnostic Information

[ctl-480.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/9335573/ctl-480.txt)

Panday-Ichihara, 2022-08-15:
I have two CTL - 480 and one CTL - 471, both ctl-480 do the same behavior, while ctl-471 works just fine. first i was thinking about performance issue, but is not the case.

jamesbt365, 2022-08-15:
From the diagnostic you have sent i can see that this is with Devocub filters. 

This is likely the fact that this filter "under performs" ever so slightly. This is probably because of the way we ported the filters originally or a thing with the original devocub driver that was ported over (devocub is weird anyway for some reason it does stuff with pressure). If you think that its the filters problem open it up on the respective repo in the org.

nue0715, 2022-09-04:
Yeah I've noticed this too, I have one CTH-480, one CTH-680, and a CTL 480. The CTL 480 is the only one that acts weird and only shows 910-940 whilst the CTH-480/680 both stay at 999-1001 consistently.

gonX, 2022-11-13:
Does this have any actual effect on cursor consistency or is this more of a "number doesn't look pretty" issue? 

Does this only happen with Devocub/Hawku filters or also original plugins like BezierInterpolator?

Fixing this seems nontrivial, but fixes or workarounds are welcome.

Panday-Ichihara, 2022-11-13:
it does, with high hz monitor u can see like it is skipping some movements, any filter with interpolation can reproduce this behavior.

Panday-Ichihara, 2022-11-13:
i had a discussion in discord long ago where i think the #dev-chat almost found the issue itself.

Panday-Ichihara, 2022-11-13:
[image]

Panday-Ichihara, 2022-11-13:
[image]

Panday-Ichihara, 2022-11-13:
i think this is all the info i have about this issue.

gonX, 2026-10-02:
This should actually be fixed with #4917

closed 2026-10-02 by gonX