OpenTabletDriver/OpenTabletDriver #2389: Area position and flipped handedness.

SwedishSubmarine, 2022-08-17:
I have my tablet flipped and have had my area set like this: 
[image]
I did this assuming this would give me more space to move my pen up and to the left, but I've realized if you want as much space left and up as possible you actually need the area set like this:
[image]
What I'm assuming is that this is because the second area position is where the area actually is on the tablet if the area weren't flipped, as that is where the actual upper left corner of the tablet is. 
Maybe this is intended and just me being silly, but in that case it just feels unintuitive to me.

jamesbt365, 2022-08-17:
If your area is upside down to the object that you are viewing it from yes its inverted, this is how flipping things work. I don't see how any other way would work.

gonX, 2022-11-05:
relevant convo: https://discord.com/channels/615607687467761684/628651971267788800/1038542320267690156

X9VoiD, 2022-12-31:
Not a bug.

closed 2026-03-14 by gonX