OpenTabletDriver/OpenTabletDriver #2390: Add support for Huion Kamvas Pro 16 (2.5K)

DennisWG, 2022-08-21:
### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Manufacturer

Huion

### Product Name

Kamvas Pro 16 (2.5K)

### Device Width

349.6

### Device Height

196.7

### Pressure Levels

8192

### Additional Information

The Huion Kamvas Pro 16 (2.5K) is not supported out-of-the-box. I got it working using the following custom configuration:

```json
{
  "Name": "Huion Kamvas Pro 16 (2.5K)",
  "Specifications": {
    "Digitizer": {
      "Width": 349.6,
      "Height": 196.7,
      "MaxX": 69926.0,
      "MaxY": 39333.0
    },
    "Pen": {
      "MaxPressure": 8191,
      "Buttons": {
        "ButtonCount": 2
      }
    },
    "AuxiliaryButtons": {
      "ButtonCount": 8
    },
    "MouseButtons": null,
    "Touch": null
  },
  "DigitizerIdentifiers": [
    {
      "VendorID": 9580,
      "ProductID": 109,
      "InputReportLength": 12,
      "OutputReportLength": null,
      "ReportParser": "OpenTabletDriver.Configurations.Parsers.Huion.GianoReportParser",
      "FeatureInitReport": null,
      "OutputInitReport": null,
      "DeviceStrings": {
        "201": "HUION_M214_\\d{6}$"
      },
      "InitializationStrings": null
    }
  ],
  "AuxilaryDeviceIdentifiers": [],
  "Attributes": {}
}
```
[... 226 lines cut ...]
      "Message": "Driver is enabled.",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    },
    {
      "Time": "2022-08-21T02:21:41.780684+02:00",
      "Group": "Display",
      "Message": "DPI Awareness enabled",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2022-08-21T02:21:41.7806881+02:00",
      "Group": "Settings",
      "Message": "Presets have been refreshed. Loaded 0 presets.",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    }
  ]
}

```

jamesbt365, 2022-08-21:
Don't round the height and width, max sure the max x and y are correct in the `tablet > tablet debugger` check if the max pressure is correct and the tilt (if the tablet has any) works correctly. Also check if the tablet buttons and the pen button works then you can make a pull request and add this configuration if you would prefer to do that.

DennisWG, 2022-08-21:
Yep, I've checked with the tablet debugger. Works fine. Also, I'm not rounding width and height. I've taken the values directly from Huion's product specification. Same with pressure for that matter.

jamesbt365, 2022-08-21:
Spec sheets are often inaccurate, the correct width and height are 349.63x196.665

jamesbt365, 2022-09-12:
fixed in #2410

closed 2022-09-12 by jamesbt365