OpenTabletDriver/OpenTabletDriver #2395: OpenTabletDriver disappears completely from the system. | Linux

noemsus, 2022-08-22:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Steps to Reproduce

When I download OpenTabletDriver in Linux Mint and when I open it, it works. But when i close it, it disappears from the system completely. 

### Expected Behavior

It should close and not remove entirely from the system. 

### Observed Behavior

I closed the application and removed entirely from the system

### Tablet Device

Wacom CTH-480

### Diagnostic Information

it's removed no folder nothing e.e

jamesbt365, 2022-08-22:
As in its deleted? This very much seems like user error.

noemsus, 2022-08-22:
> As in its deleted? This very much seems like user error.

it deleted completely idk what happened. I did the installation guide step by step and every time it removed itself.

noemsus, 2022-08-22:
on other distros I used something like that didn't happen

jamesbt365, 2022-08-22:
As otd is installed into a root directory there is no way it can just delete itself. `stat /usr/share/OpenTabletDriver`

noemsus, 2022-08-22:
it isn't there...

jamesbt365, 2022-08-22:
So install it and record it "deleting itself"

noemsus, 2022-08-22:
nwm i fixed it somehow