OpenTabletDriver/OpenTabletDriver #2396: OpenTabletDriver is detecting tablet but when touching with the pen it says "No tablets are detected." | Linux Mint

noemsus, 2022-08-22:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Steps to Reproduce

https://streamable.com/ut5tsc# 

### Expected Behavior

It shouldn't show "No tablets are detected." when pressing/dragging with the pen on the tablet.

### Observed Behavior

When you press or drag on the tablet it shows: "No tablets are detected."

### Tablet Device

CTL-480

### Diagnostic Information

{
  "App Version": "OpenTabletDriver v0.6.0.4",
  "Build Date": "05/04/2022",
  "Operating System": {
    "Platform": 4,
    "ServicePack": "",
    "Version": "5.15.0.46",
    "VersionString": "Unix 5.15.0.46"
  },
  "Environment Variables": {
    "USER": "noem",
    "TEMP": null,
    "TMP": null,
    "TMPDIR": null,
    "DISPLAY": ":0",
    "WAYLAND_DISPLAY": null,
    "PWD": "/home/noem",
    "PATH": "/home/noem/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/home/noem/.dotnet/tools"
  },
  "HID Devices": [
    {
      "DevicePath": "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10.2/1-10.2:1.5/0003:1038:1282.000F/hidraw/hidraw2",
      "Manufacturer": "SteelSeries",
      "ProductName": "SteelSeries GameDAC",
      "SerialNumber": "000000000000",
      "FriendlyName": "SteelSeries GameDAC",
      "VendorID": 4152,
      "ProductID": 4738,
      "InputReportLength": -1,
      "OutputReportLength": -1,
      "FeatureReportLength": -1,
      "CanOpen": true
    },
    {
      "DevicePath": "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10.3/1-10.3:1.0/0003:1038:1280.000C/hidraw/hidraw11",
      "Manufacturer": "SteelSeries",
      "ProductName": "SteelSeries GameDAC",
      "SerialNumber": "",
      "FriendlyName": "SteelSeries GameDAC",
      "VendorID": 4152,
      "ProductID": 4736,
      "InputReportLength": -1,
      "OutputReportLength": -1,
      "FeatureReportLength": -1,
      "CanOpen": true
[... 199 lines cut ...]
      "Message": "Searching for tablet '10moon 1060N'",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2022-08-22T16:01:57.053893+02:00",
      "Group": "Detect",
      "Message": "Searching for tablet 'Acepen AP 1060'",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2022-08-22T16:01:57.0539113+02:00",
      "Group": "Detect",
      "Message": "Searching for tablet 'Artisul A1201'",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2022-08-22T16:01:57.0539357+02:00"
[Diagnostic_Information.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/9395010/Diagnostic_Information.txt)
,

jamesbt365, 2022-08-22:
http://opentabletdriver.net/Wiki/FAQ/Linux#ghost-cursor

InfinityGhost, 2022-08-26:
Closing as there has been no update and FAQ pages provide a solution.

closed 2022-08-26 by InfinityGhost