OpenTabletDriver/OpenTabletDriver #2409: Float error in settings

Kuuuube, 2022-08-27:
## Description

With some settings, numbers cannot be input precisely. For example 159.6 cannot be input into the area and instead becomes 159.600006103515625.

[image]

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | Win10
| Software Version | [82b9c1f](https://github.com/OpenTabletDriver/OpenTabletDriver/commit/82b9c1f2131a4312cab456a3bc73b5d39a2113fb)

gonX, 2023-10-01:
Cannot reproduce on v0.6.3.0, but it does still happen on `master` (6c4ffacb9)

gonX, 2025-01-24:
Needs verification on v0.6.5.0 otherwise this is solved with Avalonia.

jamesbt365, 2025-01-24:
This was never a problem in 0.6.x, it was a post DI regression.

closed 2025-01-24 by jamesbt365