OpenTabletDriver/OpenTabletDriver #2412: Add support for UGEE EX07

CrimsonFork, 2022-08-29:
### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Manufacturer

Ugee

### Product Name

EX07

### Device Width

203mm

### Device Height

126mm

### Pressure Levels

2048

### Lines per Inch (LPI)

5080

### Additional Information

Digimend [supports this tablet](https://digimend.github.io/tablets/Ugee_EX07)

### Diagnostic Information

[ugee_ex07.json](https://github.com/OpenTabletDriver/OpenTabletDriver/files/9442900/ugee_ex07.txt)

jamesbt365, 2022-08-31:
Move [99-opentabletdriver.rules.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/9460873/99-opentabletdriver.rules.txt) to /etc/udev/rules.d/99-opentabletdriver.rules and remove the txt at the end (github won't accept .rules as a format for some reason) Then run `sudo udevadm control --reload-rules` and replug your tablet. Export another diagnostic after restarting the service for OpenTabletDriver

CrimsonFork, 2022-08-31:
[ugee_ex07_udev_rules.json](https://github.com/OpenTabletDriver/OpenTabletDriver/files/9460977/ugee_ex07_udev_rules.json.txt)

Interestingly, as soon as I ran the command the tablet started working again, though I'm not sure if it's OpenTabletDriver's doing.

CrimsonFork, 2022-08-31:
```bash
$ echo "blacklist hid_uclogic" | sudo tee -a /etc/modprobe.d/blacklist.conf
sudo rmmod hid_uclogic
blacklist hid_uclogic
rmmod: ERROR: Module hid_uclogic is not currently loaded
```

jamesbt365, 2022-08-31:
restart otd once again and export another diagnostic

CrimsonFork, 2022-08-31:
[ugee_ex07_udev_nohid_uclogic.json](https://github.com/OpenTabletDriver/OpenTabletDriver/files/9461267/ugee_ex07_udev_nohid_uclogic.json.txt)

jamesbt365, 2022-08-31:
Create the directory /usr/share/OpenTabletDriver/Configurations, move 
[this](https://github.com/OpenTabletDriver/OpenTabletDriver/files/9461287/EX07.json.txt) file into it and remove the .txt at the end, it should now detect, when it does go into the `tablet > tablet debugger` and see if there is a dot on the visualizer when moving your pen about.

CrimsonFork, 2022-08-31:
Doing that with hif_uclogic disabled spews out errors:

[image]
[image]

Although disabling it at runtime does at least make the name appear
[image]

jamesbt365, 2022-08-31:
could you get the stack trace from the daemon?

CrimsonFork, 2022-08-31:
How do I do that? I tried looking up but there seems to be too much terminology I just don't know.

jamesbt365, 2022-08-31:
get the log the daemon gives

jamesbt365, 2022-08-31:
i think i know whats wrong anyway

jamesbt365, 2022-08-31:
[EX07.json.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/9461600/EX07.json.txt)
replace the file with this one, remember to remove the .txt

CrimsonFork, 2022-08-31:
Still fails in the same way. And I'm still unsure how to get the log, as /var/log doesn't seem to contain anything opentabletdriver related.

jamesbt365, 2022-08-31:
correct way would be journalctl --user-unit opentabletdriver if you were running through the service, either way remove ~/.config/OpenTabletDriver and restart the service

CrimsonFork, 2022-08-31:
Removed the config.

```bash
$ journalctl --user-unit opentabletdriver
Aug 31 14:59:47 s systemd[2205]: opentabletdriver.service: Scheduled restart job, restart counter is at 67.
Aug 31 14:59:47 s systemd[2205]: Stopped OpenTabletDriver Daemon.
Aug 31 14:59:47 s systemd[2205]: Started OpenTabletDriver Daemon.
Aug 31 14:59:47 s OpenTabletDriver.Daemon[50859]: [Detect:Info]        Searching for tablets...
Aug 31 14:59:47 s OpenTabletDriver.Daemon[50859]: [Detect:Debug]        Configuration overrides exist in '/usr/share/OpenTabletDriver/Configurations', overriding built-in configurations.
Aug 31 14:59:47 s OpenTabletDriver.Daemon[50859]: [Detect:Debug]        Searching for tablet '10moon 1060N'
Aug 31 14:59:47 s OpenTabletDriver.Daemon[50859]: [Detect:Debug]        Searching for tablet 'Acepen AP 1060'
Aug 31 14:59:47 s OpenTabletDriver.Daemon[50859]: [Detect:Debug]        Searching for tablet 'Artisul A1201'
Aug 31 14:59:47 s OpenTabletDriver.Daemon[50859]: [Detect:Debug]        Searching for tablet 'Artisul AP604'
Aug 31 14:59:47 s OpenTabletDriver.Daemon[50859]: [Detect:Debug]        Searching for tablet 'Artisul M0610 Pro'
Aug 31 14:59:47 s OpenTabletDriver.Daemon[50859]: [Detect:Debug]        Searching for tablet 'Gaomon 1060 Pro'
Aug 31 14:59:47 s OpenTabletDriver.Daemon[50859]: [Detect:Debug]        Searching for tablet 'Gaomon M106K Pro'
Aug 31 14:59:47 s OpenTabletDriver.Daemon[50859]: [Detect:Debug]        Searching for tablet 'Gaomon M106K'
Aug 31 14:59:47 s OpenTabletDriver.Daemon[50859]: [Detect:Debug]        Searching for tablet 'Gaomon M10K Pro'
Aug 31 14:59:47 s OpenTabletDriver.Daemon[50859]: [Detect:Debug]        Searching for tablet 'Gaomon M10K'
Aug 31 14:59:47 s OpenTabletDriver.Daemon[50859]: [Detect:Debug]        Searching for tablet 'Gaomon M1220'
Aug 31 14:59:47 s OpenTabletDriver.Daemon[50859]: [Detect:Debug]        Searching for tablet 'Gaomon M1230'
Aug 31 14:59:47 s OpenTabletDriver.Daemon[50859]: [Detect:Debug]        Searching for tablet 'Gaomon M6'
Aug 31 14:59:47 s OpenTabletDriver.Daemon[50859]: [Detect:Debug]        Searching for tablet 'Gaomon PD1161'
Aug 31 14:59:47 s OpenTabletDriver.Daemon[50859]: [Detect:Debug]        Searching for tablet 'Gaomon PD1560'
Aug 31 14:59:47 s OpenTabletDriver.Daemon[50859]: [Detect:Debug]        Searching for tablet 'Gaomon PD1561'
Aug 31 14:59:47 s OpenTabletDriver.Daemon[50859]: [Detect:Debug]        Searching for tablet 'Gaomon S56K'
Aug 31 14:59:47 s OpenTabletDriver.Daemon[50859]: [Detect:Debug]        Searching for tablet 'Gaomon S620'
Aug 31 14:59:47 s OpenTabletDriver.Daemon[50859]: [Detect:Debug]        Searching for tablet 'Gaomon S630'
Aug 31 14:59:47 s OpenTabletDriver.Daemon[50859]: [Detect:Debug]        Searching for tablet 'Genius G-Pen 560'
Aug 31 14:59:47 s OpenTabletDriver.Daemon[50859]: [Detect:Debug]        Searching for tablet 'Genius i608x'
Aug 31 14:59:47 s OpenTabletDriver.Daemon[50859]: [Detect:Debug]        Searching for tablet 'Huion 420'
Aug 31 14:59:47 s OpenTabletDriver.Daemon[50859]: [Detect:Debug]        Searching for tablet 'Huion G10T'
Aug 31 14:59:47 s OpenTabletDriver.Daemon[50859]: [Detect:Debug]        Searching for tablet 'Huion GC610'
Aug 31 14:59:47 s OpenTabletDriver.Daemon[50859]: [Detect:Debug]        Searching for tablet 'Huion GT-220 V2'
Aug 31 14:59:47 s OpenTabletDriver.Daemon[50859]: [Detect:Debug]        Searching for tablet 'Huion GT-221 Pro'
Aug 31 14:59:47 s OpenTabletDriver.Daemon[50859]: [Detect:Debug]        Searching for tablet 'Huion H1060P'
Aug 31 14:59:47 s OpenTabletDriver.Daemon[50859]: [Detect:Debug]        Searching for tablet 'Huion H1161'
Aug 31 14:59:47 s OpenTabletDriver.Daemon[50859]: [Detect:Debug]        Searching for tablet 'Huion H420'
Aug 31 14:59:47 s OpenTabletDriver.Daemon[50859]: [Detect:Debug]        Searching for tablet 'Huion H420X'
Aug 31 14:59:47 s OpenTabletDriver.Daemon[50859]: [Detect:Debug]        Searching for tablet 'Huion H430P'
Aug 31 14:59:47 s OpenTabletDriver.Daemon[50859]: [Detect:Debug]        Searching for tablet 'Huion H580X'
Aug 31 14:59:47 s OpenTabletDriver.Daemon[50859]: [Detect:Debug]        Searching for tablet 'Huion H610 Pro V2'
Aug 31 14:59:47 s OpenTabletDriver.Daemon[50859]: [Detect:Debug]        Searching for tablet 'Huion H610 Pro'
Aug 31 14:59:47 s OpenTabletDriver.Daemon[50859]: [Detect:Debug]        Searching for tablet 'Huion H610X'
Aug 31 14:59:47 s OpenTabletDriver.Daemon[50859]: [Detect:Debug]        Searching for tablet 'Huion H640P'
Aug 31 14:59:47 s OpenTabletDriver.Daemon[50859]: [Detect:Debug]        Searching for tablet 'Huion H642'
Aug 31 14:59:47 s OpenTabletDriver.Daemon[50859]: [Detect:Debug]        Searching for tablet 'Huion H950P'
Aug 31 14:59:47 s OpenTabletDriver.Daemon[50859]: [Detect:Debug]        Searching for tablet 'Huion HC16'
Aug 31 14:59:47 s OpenTabletDriver.Daemon[50859]: [Detect:Debug]        Searching for tablet 'Huion HS610'
Aug 31 14:59:47 s OpenTabletDriver.Daemon[50859]: [Detect:Debug]        Searching for tablet 'Huion HS611'
Aug 31 14:59:47 s OpenTabletDriver.Daemon[50859]: [Detect:Debug]        Searching for tablet 'Huion HS64'
Aug 31 14:59:47 s OpenTabletDriver.Daemon[50859]: [Detect:Debug]        Searching for tablet 'Huion HS95'
Aug 31 14:59:47 s OpenTabletDriver.Daemon[50859]: [Detect:Debug]        Searching for tablet 'Huion Kamvas 13'
Aug 31 14:59:47 s OpenTabletDriver.Daemon[50859]: [Detect:Debug]        Searching for tablet 'Huion Kamvas 16 (2021)'
[... cut ...]
Aug 31 14:59:48 s OpenTabletDriver.Daemon[50859]:    at System.Threading.ThreadPoolWorkQueue.Dispatch()
Aug 31 14:59:48 s OpenTabletDriver.Daemon[50859]:    at System.Threading.PortableThreadPool.WorkerThread.WorkerThreadStart()
Aug 31 14:59:48 s OpenTabletDriver.Daemon[50859]:    at System.Threading.Thread.StartCallback()
Aug 31 14:59:49 s systemd[2205]: opentabletdriver.service: Main process exited, code=killed, status=6/ABRT
Aug 31 14:59:49 s systemd[2205]: opentabletdriver.service: Failed with result 'signal'.
```

jamesbt365, 2022-08-31:
Your environment is a little messed up i think but we can fix that after, for now stop the systemd user server and start it up manually with `dotnet /usr/share/OpenTabletDriver/OpenTabletDriver.Daemon.dll`

CrimsonFork, 2022-08-31:
It works now, or at least it functions in that it doesn't crash and does show up in the debugger

CrimsonFork, 2022-08-31:
Yes. `Position:[40640,25400]`

jamesbt365, 2022-08-31:
max pressure when pressing the pen nib down?

CrimsonFork, 2022-08-31:
2047

jamesbt365, 2022-08-31:
I can't find a spec sheet but does this tablet have anything special like tilt, etc etc.

CrimsonFork, 2022-08-31:
There's also a bug of some sort when the pen is approached from outside the zone from the top or right, or pressing a pen button while outside, making the pointer jump at almost the opposite edge until it's within the zone again, but that's not specific to OpenTabletDriver and I'm not sure how easy that would be to fix in software.

jamesbt365, 2022-08-31:
[EX07.json.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/9461919/EX07.json.txt)
This configuration _might_ fix the buttons (i hope i don't need to change a parser) but about the teleportation while pressing pen buttons, yeah that can't be fixed in software.

CrimsonFork, 2022-08-31:
Nope, it still crashes the thing
`(OpenTabletDriver.UX.Gtk:74718): Gtk-WARNING **: 15:36:13.986: __gtksharp_2_Eto_GtkSharp_Forms_EtoEventBox 0x562fb8aa6130 adjusted size vertical min 320 natural 320 must not decrease below min 343 natural 343`

```
[Tablet:Debug]	Tablet debugging is enabled
[IndexOutOfRangeException:Error]	Index was outside the bounds of the array.
   at OpenTabletDriver.Plugin.Tablet.TabletReport..ctor(Byte[] report)
   at OpenTabletDriver.Configurations.Parsers.XP_Pen.XP_PenReportParser.Parse(Byte[] report)
   at OpenTabletDriver.Devices.DeviceReader`1.Main()
[Tablet:Debug]	Tablet debugging is disabled
[IPC:Debug]	Disconnected from a client.

It only happens while it's outside the main area. it also goes to very specific coordinates, for horizontal tra

CrimsonFork, 2022-08-31:
> but about the teleportation while pressing pen buttons, yeah that can't be fixed in software.

It only happens while it's outside the main area. it also goes to very specific coordinates, for horizontal travel there's just one X value and for vertical there's a very narrow range of Y values, so I though maybe it can be easily ignored when form one input in a specific X or Y range it jumps to another specified range.

jamesbt365, 2022-08-31:
data is missing when the pen buttons are pressed, so you'll either get cursor teleportation or freezing entirely.

jamesbt365, 2022-08-31:
As to it crashing on pressing aux, somebody who knows how to parse it will have to continue on from here.

CrimsonFork, 2022-08-31:
It's much less of an issue when pressing the buttons and more that it's just something that happens when the pen is getting out of the designated field but is still picked up. Thank you for your help so far anyway.