OpenTabletDriver/OpenTabletDriver #2416: Monitor mapping under wayland shows different behaviours in absolute and artist mode

himekifee, 2022-09-02:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Steps to Reproduce

Switch between absolute and artist mode.

### Expected Behavior

Same mapping.

### Observed Behavior

Under absolute mode, everything works as expected. After switching to artist mode, there seems to be some issue with screen size detection. e.g. I have two monitors, both 2k. Ideally, I only need to set the width to 2560 and height to 1440 and position 1280x1280 to cover the first monitor.
[image]
It shows OK on the GUI, but when moving the cursor around, there is a margin limiting the area on the first monitor. I have to set it to the total display area(2560+1440x2560=4000x2560) to be able to use the entire first monitor
[image]

### Tablet Device

Wacom PTH-651

### Diagnostic Information

{
  "App Version": "OpenTabletDriver v0.6.0.4",
  "Build Date": "2022/7/19",
  "Operating System": {
    "Platform": 4,
    "ServicePack": "",
    "Version": "5.19.5.1",
    "VersionString": "Unix 5.19.5.1"
  },
  "Environment Variables": {
    "USER": "grider",
    "TEMP": null,
    "TMP": null,
    "TMPDIR": null,
    "DISPLAY": ":1",
    "WAYLAND_DISPLAY": "wayland-0",
    "PWD": "/usr/share/OpenTabletDriver",
    "PATH": "hidden"
  },
  "HID Devices": [
    {
      "DevicePath": "/sys/devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:08:00.3/usb3/3-6/3-6.1/3-6.1:1.0/0003:046D:C539.0007/hidraw/hidraw6",
      "Manufacturer": "Logitech",
      "ProductName": "USB Receiver",
      "SerialNumber": "",
      "FriendlyName": "USB Receiver",
      "VendorID": 1133,
      "ProductID": 50489,
      "InputReportLength": 9,
      "OutputReportLength": 2,
      "FeatureReportLength": 0,
      "CanOpen": true
    },
    {
      "DevicePath": "/sys/devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:08:00.3/usb3/3-6/3-6.3/3-6.3:1.0/0003:056A:0315.000E/hidraw/hidraw12",
      "Manufacturer": "Wacom Co.,Ltd.",
      "ProductName": "Intuos5 touch M",
      "SerialNumber": "",
      "FriendlyName": "Intuos5 touch M",
      "VendorID": 1386,
      "ProductID": 789,
      "InputReportLength": 10,
[... 1539 lines cut ...]
      "Time": "2022-09-02T01:19:30.254165+01:00",
      "Group": "Wacom PTH-651",
      "Message": "Ignoring reports outside area: Disabled",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    },
    {
      "Time": "2022-09-02T01:19:30.2541744+01:00",
      "Group": "Wacom PTH-651",
      "Message": "Tip Binding: [Mouse Button Binding: Left]@0%",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    },
    {
      "Time": "2022-09-02T01:19:30.2541839+01:00",
      "Group": "Settings",
      "Message": "Driver is enabled.",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    }
  ]
}

jamesbt365, 2022-09-02:
This seems like a problem with the way that your wayland compositor does screen mapping and i don't know if there is anything we can do for this

himekifee, 2022-09-02:
FYI it's kwin. Quite annoying as only artist mode works in OSU. Absolute mode stops moving cursor in the window as I found out [here](https://www.reddit.com/r/linux_gaming/comments/slc0ev/comment/i7glgd0/?utm_source=share&utm_medium=web2x&context=3).

jamesbt365, 2022-09-02:
Duplicate of #1858

gonX, 2022-09-02:
Absolute mode not working correctly on XWayland (or at least in osu!) is known by a couple of projects members, but not currently tracked. I'll test it a bit more and then make an issue for it.

closed 2022-09-02 by gonX

himekifee, 2022-10-14:
> Absolute mode not working correctly on XWayland (or at least in osu!) is known by a couple of projects members, but not currently tracked. I'll test it a bit more and then make an issue for it.

Any update on this? Have been AFK osu for 2 months.

jamesbt365, 2022-10-14:
No changes.

himekifee, 2022-10-14:
Any hint that I can try to debug?

jamesbt365, 2022-10-14:
unsure because (i think) it is not an OTD issue.

gonX, 2022-10-15:
I tried reproducing the issue today and it seems to have fixed itself. I'm guessing this is a libinput or DE issue that has been resolved in a newer version on my end.

If you're not on a rolling release distro it might take a while for the next version to be pushed to your distro.

himekifee, 2022-10-15:
I'm on latest Archlinux. It's still happening to me.

InfinityGhost, 2022-10-15:
It is still not an OTD issue