OpenTabletDriver/OpenTabletDriver #2418: Proposal: Move configuration metadata into individual configuration file

InfinityGhost, 2022-09-03:
Currently, tablet metadata such as support status, requirements, and workarounds are all stored in `TABLETS.md`. Moving all of this information into the configuration file itself makes it easier to maintain and would no longer be bound to a single file that is prone to formatting issues.

As for https://opentabletdriver.net, we can render the tablets listing from the configurations files themselves, as we already pull and cache the entire repository for other functions in the site.

gonX, 2022-11-05:
Support!

Doing this automatically seems trivial, I'll probably give it a shot if I have some spare time the next couple of days

jamesbt365, 2022-11-05:
#2437

jamesbt365, 2025-01-24:
Is this still wanted? Attempts are stale.