OpenTabletDriver/OpenTabletDriver #2420: OTD no longer detects Tablet

RaijinXBlade, 2022-09-03:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Steps to Reproduce

I noticed there was no way to set the pen/tablet buttons and get them to actually register, nor was there pen pressure sensitivity.
I saw something about needing to use Windows Ink through OTD.
I've followed the VMulti/Windows Ink instructions, and now my Tablet is no longer detected in OTD.
I uninstalled VMulti and restarted, and still nothing is detected even though it was before I followed the VMulti steps.

### Expected Behavior

Opening OTD should detect tablet, evidently.

### Observed Behavior

Covered in the above steps. Worth noting that the tablet is indeed plugged in and recognised by Windows.

Additionally, I'd appreciate any help resolving this, and ideally some implementation that doesn't require Windows Ink to be able to use this thing with pressure sensitivity and mappable buttons, or at the very least, a more elegant solution than the VMulti requirement.

### Tablet Device

Wacom CTH-680

### Diagnostic Information

[Detect:Warning]	Another tablet driver found: Wacom
[Detect:Info]	Searching for tablets...
[NullReferenceException:Error]	Object reference not set to an instance of an object.
   at OpenTabletDriver.Driver.MatchDevice(TabletConfiguration config, IList`1 identifiers)
   at OpenTabletDriver.Driver.Match(TabletConfiguration config)
[Detect:Info]	No tablets were detected.
[Settings:Info]	Presets have been refreshed. Loaded 0 presets.
[Detect:Info]	Searching for tablets...
[NullReferenceException:Error]	Object reference not set to an instance of an object.
   at OpenTabletDriver.Driver.MatchDevice(TabletConfiguration config, IList`1 identifiers)
   at OpenTabletDriver.Driver.Match(TabletConfiguration config)
[Detect:Info]	No tablets were detected.

jamesbt365, 2022-09-03:
Remove %localappdata%\OpenTabletDriver and then follow the guides for said plugins correctly.

closed 2022-09-03 by jamesbt365

RaijinXBlade, 2022-09-03:
I followed the steps exactly. For the sake of it, I deleted the appdata folder and followed the steps again. Same result.

jamesbt365, 2022-09-03:
Without a proper diagnostic file, i cannot diagnose a problem and as you didn't give a diagnostic file, i cannot see what you did.

RaijinXBlade, 2022-09-03:
Sorry. If there is a diagnostic file dump somewhere, I didn't see it. If it was in the appdata folder, it's gone now. I thought the console information was that.

jamesbt365, 2022-09-03:
No, help > export diagnostics save that file as a txt and send it here as an attachment, as it directs in the template

RaijinXBlade, 2022-09-03:
[diag.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/9483537/diag.txt)

jamesbt365, 2022-09-03:
Whats in the Configurations folder you have?

RaijinXBlade, 2022-09-03:
Oh that was when I first installed OTD, and I was looking around. I thought setting the sensitivity and such in the configuration editor would enable those. This was just before I learned about the Windows Ink solution.

jamesbt365, 2022-09-03:
Remove the folder

jamesbt365, 2022-09-03:
Then restart otd.

RaijinXBlade, 2022-09-03:
Yep... Just did and it's detecting the tablet again. Sorry for the trouble.