OpenTabletDriver/OpenTabletDriver #2422: Support Request: Windows Ink on OpenTabletDriver still doesn't work after installing VMultiDriver

AwesomeGamer89, 2022-09-05:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### What do you need help with?

The pen tip does not do anything when pressed down.

### What have you tried?

I tried installing VMultiDriver after seeing this issue: https://github.com/OpenTabletDriver/OpenTabletDriver/discussions/1239 and I can move my cursor just fine, but the pen tip still does not work (left click bound).

### Diagnostic Information

[Diagnostics.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/9492059/Diagnostics.zip)

(Exceeded 65535 characters)

jamesbt365, 2022-09-05:
Please follow the installation guide for windows ink on the windows ink wiki, this is user error

closed 2022-09-05 by jamesbt365

Doksha, 2022-10-13:
because devs refuse to actually answer the question, the solution is to go into pen settings select the three dots on tip binding, set type to windows ink, and button to pen tip.