OpenTabletDriver/OpenTabletDriver #2423: Fails to load with XP-Pen Artist 16 Pro

ThatOneCalculator, 2022-09-08:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Steps to Reproduce

1. `yay -S opentabletdriver`
2. Open GUI
3. Run `otd` in terminal
4. Observe

### Expected Behavior

Works

### Observed Behavior

GUI crashed the moment `otd` is run:

```
System.Reflection.ReflectionTypeLoadException: Unable to load one or more of the requested types.
Could not load file or assembly 'OpenTabletDriver.Plugin, Version=0.6.0.0, Culture=neutral, PublicKeyToken=null'. The system cannot find the file specified.

   at System.Reflection.RuntimeModule.GetTypes(RuntimeModule module)
   at System.Reflection.RuntimeAssembly.get_DefinedTypes()
   at OpenTabletDriver.Desktop.Reflection.PluginManager.<>c.<get_PluginTypes>b__22_0(Assembly asm)
   at System.Linq.Enumerable.SelectManyIterator[TSource,TCollection,TResult](IEnumerable`1 source, Func`2 collectionSelector, Func`3 resultSelector)+MoveNext()
   at System.Linq.Enumerable.WhereSelectEnumerableIterator`2.MoveNext()
   at System.Linq.Enumerable.WhereEnumerableIterator`1.ToArray()
   at System.Linq.Enumerable.ToArray[TSource](IEnumerable`1 source)
   at System.Collections.Immutable.ImmutableArray.CreateRange[T](IEnumerable`1 items)
   at System.Collections.Immutable.ImmutableArray.ToImmutableArray[TSource](IEnumerable`1 items)
   at OpenTabletDriver.UX.Controls.OutputModePicker.CreateDefaultDataStore()
   at OpenTabletDriver.UX.Controls.OutputModePicker.<Refresh>b__3_0()
   at Eto.GtkSharp.Forms.ApplicationHandler.<>c__DisplayClass20_0.<AsyncInvoke>b__0(Object <p0>, EventArgs <p1>)
   at Gtk.Application.InvokeCB.Invoke()
   at GLib.Timeout.TimeoutProxy.Handler()
System.IO.FileNotFoundException: Could not load file or assembly 'OpenTabletDriver.Plugin, Version=0.6.0.0, Culture=neutral, PublicKeyToken=null'. The system cannot find the file specified.

File name: 'OpenTabletDriver.Plugin, Version=0.6.0.0, Culture=neutral, PublicKeyToken=null'
```

### Tablet Device

XP-Pen Artist 16 Pro

### Diagnostic Information

Unable to load

jamesbt365, 2022-09-08:
the git versions of opentabletdriver don't have plugin support yet. Use  opentabletdriver, not opentabletdriver-git.

closed 2022-09-08 by jamesbt365

ThatOneCalculator, 2022-09-08:
I tried with the regular version and got the same error, and I don't have any plugins enabled.

jamesbt365, 2022-09-08:
Restart the daemon and remove ~/.config/OpenTabletDriver. Backporting settings is not supported.