OpenTabletDriver/OpenTabletDriver #2427: windows ink tilt and rotatiopn

alkeryn, 2022-09-15:
### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Description

hey would it be possible to add tilt and rotation support to windows ink ?
i know it relies on the vmulti driver (which by default don't support tilt) and i myself modified it to add tilt support, however, i don't have signing certificates
i'd be more than willing to give you my fork of vmulti if in exchange you can sign and release it !

thank you very much !

jamesbt365, 2022-09-15:
It has tilt and rotation support.

https://github.com/X9VoiD/VoiDPlugins/wiki/WindowsInk

closed 2022-09-15 by jamesbt365

alkeryn, 2022-09-15:
realy ?
i saw no mention of it in the plugin code :thinking: 
[image]

jamesbt365, 2022-09-15:
[image]
it very much has tilt

alkeryn, 2022-09-15:
@jamesbt365 thanks !
what file is it though ?