OpenTabletDriver/OpenTabletDriver #2432: OpenTabletDriver doesn't see Huion 1060Plus

arlecchinosimp, 2022-09-23:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### What do you need help with?

OpenTabletDriver doesn't see Huion 1060Plus

### What have you tried?

I tried to install zadig on interface1, reinstall drivers, reinstall OpenTabletDriver

### Diagnostic Information

[diagnostics.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/9634442/diagnostics.txt)

jamesbt365, 2022-09-23:
Uninstall the huion driver

arlecchinosimp, 2022-09-23:
> Uninstall the huion driver

Uninstalled, still not detecting

jamesbt365, 2022-09-23:
Uninstall zadig as well, you aren't meant to have that with this tablet.

If it still doesn't work after get a string dump with 9580 in the top box and 110 in the middle and a string index of 200. If it isn't HUION_T174_6numbers then something is actually wrong.

Closing now due to user error.

closed 2022-09-23 by jamesbt365

arlecchinosimp, 2022-09-23:
How to get that string? I didn't know how to do that

jamesbt365, 2022-09-23:
tablet > device string reader.

arlecchinosimp, 2022-09-23:
Index 1: HUION
Index 2: PenTablet
Index 3: LuoYong70558
Index 200: 왰؀|῿Ꮨః
Index 201: HUION_T167_180808
Index 202: HUION Animation Technology Co.,ltd

jamesbt365, 2022-09-23:
All is fine there, it matches a tablet, uninstall zadig and you're good to go.

arlecchinosimp, 2022-09-23:
[image: image.png]
I already deleted it, have everywhere HIDusb how it was, but still
OpenDriver didn't want to work properly on my laptop, idk what to do. i'm
really sorry for wasting your time...

jamesbt365, 2022-09-23:
Care to screenshot that for me?

arlecchinosimp, 2022-09-23:
[image: image.png]
also deleted in download folder/in windows/inf folder too

jamesbt365, 2022-09-23:
You didn't upload an image.

arlecchinosimp, 2022-09-23:
[image]

arlecchinosimp, 2022-09-23:
or what should i to upload?

jamesbt365, 2022-09-23:
view > device by container > find your tablet > open the container > screenshot

arlecchinosimp, 2022-09-23:
[image]
prolly that it is

arlecchinosimp, 2022-09-23:
[image]

jamesbt365, 2022-09-23:
Sending that twice is not following the instructions, go to view > device by container > find your tablet > open the container > screenshot.

arlecchinosimp, 2022-09-23:
[image]

jamesbt365, 2022-09-23:
Again, you didn't read. I never asked for ANY right clicking.
[image]

arlecchinosimp, 2022-09-23:
[image]
like this? im so freaking stupid, my bad

jamesbt365, 2022-09-23:
help > export diagnostic, save it with a txt extension and upload it here.

arlecchinosimp, 2022-09-23:
[diagnostics2.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/9637166/diagnostics2.txt)

jamesbt365, 2022-09-23:
Fixed in master.

arlecchinosimp, 2022-09-23:
> Fixed in master.
What should i do next? I'm dumb

Sublimelime, 2022-09-23:
Simply wait for the next release. Alternatively, build OTD yourself from the master branch, there's info on how to do this in our contributing.md file.

arlecchinosimp, 2022-09-23:
Ok, thank you.