OpenTabletDriver/OpenTabletDriver #2435: Add support for Huion1060Pro

radudek332, 2022-09-23:
### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Manufacturer

Huion

### Product Name

1060Pro+

### Device Width

10'

### Device Height

6.25'

### Pressure Levels

2048

### Lines per Inch (LPI)

5080

### Diagnostic Information

```json
{
  "App Version": "OpenTabletDriver v0.6.0.4",
  "Build Date": "5/4/2022",
  "Operating System": {
    "Platform": 2,
    "ServicePack": "",
    "Version": "10.0.19044.0",
    "VersionString": "Microsoft Windows NT 10.0.19044.0"
  },
  "Environment Variables": {
    "USER": null,
    "TEMP": "C:\\Users\\Retr0\\AppData\\Local\\Temp",
    "TMP": "C:\\Users\\Retr0\\AppData\\Local\\Temp",
    "TMPDIR": null,
    "USERPROFILE": "C:\\Users\\Retr0"
  },
  "HID Devices": [
    {
      "DevicePath": "\\\\?\\hid#vid_0e6a&pid_02c0&mi_01&col05#7&19a2788&0&0004#{4d1e55b2-f16f-11cf-88cb-001111000030}\\kbd",
      "Manufacturer": "dakai",
      "ProductName": "PS/2+USB Keyboard",
      "SerialNumber": "",
      "FriendlyName": "PS/2+USB Keyboard",
      "VendorID": 3690,
      "ProductID": 704,
      "InputReportLength": 8,
      "OutputReportLength": 0,
      "FeatureReportLength": 0,
      "CanOpen": false
    },
    {
      "DevicePath": "\\\\?\\hid#tablethid&col05#1&2d595ca7&1&0004#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "Manufacturer": "DJP Inc.",
      "ProductName": "Virtual Multitouch Device",
      "SerialNumber": "123123123",
      "FriendlyName": "Virtual Multitouch Device",
      "VendorID": 255,
      "ProductID": 47820,
[... 1805 lines cut ...]
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2022-09-23T23:09:33.5645501+02:00",
      "Group": "Display",
      "Message": "DPI Awareness enabled",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2022-09-23T23:09:33.5645657+02:00",
      "Group": "Settings",
      "Message": "Presets have been refreshed. Loaded 0 presets.",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    }
  ]
}

```
[diagnostics.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/9637123/diagnostics.txt)

jamesbt365, 2022-09-23:
Uninstall huion drivers, after you do that tablet > device string reader
9580
110
nothing
dump all.
Save that file and send it here.

radudek332, 2022-09-23:
[String Dump.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/9637147/String.Dump.txt)

jamesbt365, 2022-09-23:
Close OTD, install the OEM drivers then follow this:

https://desowin.org/usbpcap/thankyou.html?file=1.5.4.0/USBPcapSetup-1.5.4.0.exe Download and install this tool

- When done, head to C:\Program Files\USBPcap in the explorer (you can probably copy paste that path in the path bar)

- Once you're there, start USBPcapCMD.exe

- A console will appear, answer n at the first prompt and reboot your pc

- Head back to C:\Program  Files\USBPcap,

- Start USBPcapCMD.exe,

- Screenshot

radudek332, 2022-09-23:
OEM drivers as in the huion ones? Just making sure

jamesbt365, 2022-09-23:
as OEM is Original Equipment Manufacturer, yes.

radudek332, 2022-09-23:
[image]

radudek332, 2022-09-23:
I think a bit might've gotten cut off...

radudek332, 2022-09-23:
[image]
Yep, a bit was cut off, here's the entirety