OpenTabletDriver/OpenTabletDriver #2441: Relative mode broken after DI

jamesbt365, 2022-09-30:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Steps to Reproduce

1) Open OpenTabletDriver
2) Switch to the relative mode output mode
3) defaults are all 0, which means it doesn't work.
4) Relative mode settings will only save ONCE each time the daemon is started meaning you have to restart whenever you want to change settings.
5) [error](https://user-images.githubusercontent.com/22662856/193264834-95ff10a0-9413-4f09-9433-9ceeec72ed08.png) is thrown when OpenTabletDriver is restarted and set to relative mode and then the [settings stay blank](https://user-images.githubusercontent.com/22662856/193265220-b2477b0a-2ce9-4826-8399-d728478c8bae.png) until the mode is changed to another one and back to relative.

### Expected Behavior

Works just as 0.6.0.4 does.

### Observed Behavior

Explained and showed above in the steps to reproduce

### Tablet Device

N/A

### Diagnostic Information

shouldn't be needed but if needed i'll export.

jamesbt365, 2022-09-30:
Also restarting the daemon is this state causes the ui to lockup and not render anything.

InfinityGhost, 2022-09-30:
Works on my machine ™️

jamesbt365, 2022-09-30:
your machine isn't always right :P

https://discord.com/channels/615607687467761684/723399565780189234/1025368041212620830

gonX, 2023-03-28:
Can reproduce on sway (Wayland) 961578a2, however I can change the bindings without any issue. Only way to change relative mode sensitivity is by first changing to another mode and applying, and then going back to relative mode with the new desired settings.

gonX, 2025-01-24:
Fixed on avalonia branch

closed 2025-01-24 by gonX