OpenTabletDriver/OpenTabletDriver #2443: Cursor position is off in KDE apps

janTaja, 2022-10-01:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Steps to Reproduce

1. Open a KDE app, for example Krita
2. Attempt to interact with the GUI using the tablet

### Expected Behavior

Interface elements under the cursor get interacted with

### Observed Behavior

Instead, the cursor position is always off, or rather the app thinks the cursor is in a different position than it actually is. The distance and direction can change between app launches (but it stays the same in all apps). The fake cursor position seems to always be to the right and/or bottom from the real one, though.

The problem doesn't occure while the cursor is on Krita's canvas, only while it's outside of it.

Here's a bad recording showing off the problem: Krita thinks the cursor is far to the right from where it actually is, making it almost unusable. This is about the maximum distance it gets, most of the time this fake cursor is a button's height or two below and sometimes to the right to of the real cursor.

https://user-images.githubusercontent.com/53832230/193427080-ae50b20b-81f4-42b3-bda6-46103261e389.mp4

I used to have this problem with all KDE apps I've used, while I was on KDE (and didn't realize I could report it); now I've switched to GNOME and only have it with Krita and not KDE's System Settings, for example. I don't have the knowledge necessary to explain that...

### Tablet Device

Gaomon S620

### Diagnostic Information

```json
{
  "App Version": "OpenTabletDriver v0.6.0.4",
  "Build Date": "05/04/2022",
  "Operating System": {
    "Platform": 4,
    "ServicePack": "",
    "Version": "5.19.0.76051900",
    "VersionString": "Unix 5.19.0.76051900"
  },
  "Environment Variables": {
    "USER": "slava",
    "TEMP": null,
    "TMP": null,
    "TMPDIR": null,
    "DISPLAY": ":1",
    "WAYLAND_DISPLAY": null,
    "PWD": "/home/slava",
    "PATH": "/home/slava/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/home/slava/.dotnet/tools"
  },
  "HID Devices": [
    {
      "DevicePath": "/sys/devices/pci0000:00/0000:00:08.1/0000:05:00.3/usb1/1-2/1-2.4/1-2.4:1.0/0003:256C:006D.0004/hidraw/hidraw1",
      "Manufacturer": "GAOMON",
      "ProductName": "Gaomon Tablet",
      "SerialNumber": "",
      "FriendlyName": "Gaomon Tablet",
      "VendorID": 9580,
      "ProductID": 109,
      "InputReportLength": 12,
      "OutputReportLength": 0,
      "FeatureReportLength": 0,
      "CanOpen": true
    },
    {
      "DevicePath": "/sys/devices/pci0000:00/0000:00:08.1/0000:05:00.3/usb1/1-2/1-2.3/1-2.3:1.0/0003:046D:C534.0003/hidraw/hidraw2",
[... 1892 lines cut ...]
      "Message": "Disconnected from a client.",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2022-10-01T22:22:51.9015879+02:00",
      "Group": "IPC",
      "Message": "Connected to a client.",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2022-10-01T22:22:51.9015998+02:00",
      "Group": "Settings",
      "Message": "Presets have been refreshed. Loaded 0 presets.",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    }
  ]
}

```

jamesbt365, 2022-10-01:
As this is specific applications I very highly doubt that is to do with anything otd related.

Kiamrt, 2022-10-03:
I have a similar bug on kde too, except it is not only in kde applications(?) it just cuts off at the sides of the screen

jamesbt365, 2022-10-03:
wayland?
#1858

Kiamrt, 2022-10-03:
nope, X11 i am the same guy that asked for help on the discord support channel

Sibusten, 2022-11-01:
Also having this issue in Krita. Interestingly, it only appears to happen when using OTD in "Artist Mode". "Absolute Mode" works fine, but no longer has pressure sensitivity...

Using KDE and x11, tablet is Wacom CTH-670

Sibusten, 2022-11-01:
Further information from some experimenting:
It appears the magnitude of the offset depends on where the window is positioned on the screen. The further it is to the right, the further the offset of the cursor to the right, and same for vertical. Window size does not affect this.

Another note is that I am using a 4k monitor with UI scaling. Looking through the OP's logs, I can see they are likely using UI scaling as well (`Display area: [3408x2130@<5544, 1065>:0°]`). Perhaps there is some issue with calculating the window's position when UI scaling is enabled, for some applications?

AkiSakurai, 2022-11-01:
#2083

jamesbt365, 2022-11-01:
X11, not wayland.

Sibusten, 2022-11-02:
Just confirmed, resetting `Global scale` (Under `Display Configuration`) back to 100% causes the problem to go away for me. Of course I can't just turn off scaling, so it's not a great workaround 😄

gonX, 2022-11-02:
@Sibusten thanks, this is very useful information. I think we've found the culprit.

It seems that the global scale option messes with the resolution reported to OTD, which causes some inconsistencies.

Please do this for us in the following order
- Re-enable global scale (and restart computer, or relog, as necessary)
- Restart OTD if it was not restarted (restarting computer does this)
- Upload a diagnostics report from OTD (help -> export diagnostics)
- Tell us your monitor resolution(s)

Sibusten, 2022-11-02:
Logs:
[diagnostics.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/9915014/diagnostics.txt)

<details>
<summary>Monitors and OTD setup</summary>

[image]
[image]
</details>

AkiSakurai, 2022-11-02:
https://github.com/qt/qtbase/commit/9d51fb579bb4655f6740096f17f1ced50258c28f
Seems it is bug in qt, fixed in qt6.

Sibusten, 2022-11-02:
Thanks for tracking that down! That's unfortunate, because Krita is stuck on qt5. It seems there's nothing to be done for now besides avoid using UI scaling with qt5 apps.

I've managed to disable scaling for just Krita by launching it like this: `QT_SCREEN_SCALE_FACTORS=1 krita`. ([reference](https://doc.qt.io/qt-5/highdpi.html#high-dpi-support-in-qt))

It's a bit small, but manageable.

Since this is not an issue with OTD, I believe this can be closed now.

gonX, 2022-11-03:
Closing as wontfix since this seems to be a qt5 issue, and workarounds exist.

We're using the recommended methods to gather screen coordinates likewise for setting up the tablet coordinates, so I usually assume user or external software issue whenever Artist Mode is broken. Thanks for confirming this.

As a side note: it would surprise me if Krita's bugtracker does not have the issue reported.

closed 2022-11-03 by gonX