OpenTabletDriver/OpenTabletDriver #2451: Couldn't recognize Veikk A15

NeekoVN, 2022-10-12:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Steps to Reproduce

1. Plug in the tablet
2. Open OpenTabletDriver
3. "No tablets were detected"

### Expected Behavior

It should have detected the Veikk A15 as it is supported on your page

### Observed Behavior

No tablets were detected

### Tablet Device

Veikk A15

### Diagnostic Information

```json
{
  "App Version": "OpenTabletDriver v0.6.0.4",
  "Build Date": "5/4/2022",
  "Operating System": {
    "Platform": 2,
    "ServicePack": "",
    "Version": "10.0.19044.0",
    "VersionString": "Microsoft Windows NT 10.0.19044.0"
  },
  "Environment Variables": {
    "USER": null,
    "TEMP": "C:\\Users\\Admin\\AppData\\Local\\Temp",
    "TMP": "C:\\Users\\Admin\\AppData\\Local\\Temp",
    "TMPDIR": null,
    "USERPROFILE": "C:\\Users\\Admin"
  },
  "HID Devices": [
    {
      "DevicePath": "\\\\?\\hid#vid_25a7&pid_fa70&mi_01&col04#7&39cfd03f&0&0003#{4d1e55b2-f16f-11cf-88cb-001111000030}\\kbd",
      "Manufacturer": "Compx",
      "ProductName": "2.4G Wireless Receiver",
      "SerialNumber": "",
      "FriendlyName": "2.4G Wireless Receiver",
      "VendorID": 9639,
      "ProductID": 64112,
      "InputReportLength": 8,
      "OutputReportLength": 0,
      "FeatureReportLength": 0,
      "CanOpen": false
    },
    {
      "DevicePath": "\\\\?\\hid#vid_05ac&pid_0252&mi_01&col01#9&286e8e30&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "Manufacturer": "Apple Inc.",
      "ProductName": "Touchpad",
      "SerialNumber": "",
      "FriendlyName": "Touchpad",
      "VendorID": 1452,
      "ProductID": 594,
      "InputReportLength": 6,
      "OutputReportLength": 0,
      "FeatureReportLength": 0,
      "CanOpen": false
[... 1797 lines cut ...]
      "Message": "Connected to a client.",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2022-10-12T13:20:57.8996846+07:00",
      "Group": "Display",
      "Message": "DPI Awareness enabled",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2022-10-12T13:20:57.8996938+07:00",
      "Group": "Settings",
      "Message": "Presets have been refreshed. Loaded 0 presets.",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    }
  ]
}

```

jamesbt365, 2022-10-12:
Make a `Configurations` folder next to the executables of otd, move this file there and remove the .txt at the end. [A15.json.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/9762835/A15.json.txt)

Replug the tablet once done, go into the debugger:
- get the max x and y coordinates from moving your pen to the bottom right of the tablet.
- Check if the pen buttons work.
- Check the max pressure when pressing the pen nib down.

jamesbt365, 2022-10-12:
Fixed in #2417

closed 2022-10-12 by jamesbt365