OpenTabletDriver/OpenTabletDriver #2457: CTF-430: GetToolReport treats data as DeviceReport rather than IntuosTabletReport

exys228, 2022-10-16:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Steps to Reproduce

1. Connect Wacom CTF-430 tablet to the PC
2. Start OpenTabletDriver
3. Notice that mouse cursor is not moving while Raw Data in Tablet Debugger windows is being reported

### Expected Behavior

Mouse cursor should've moved

### Observed Behavior

Mouse cursor didn't move

### Tablet Device

Wacom CTF-430

### Diagnostic Information

```json
{
  "App Version": "OpenTabletDriver v0.6.0.4",
  "Build Date": "5/4/2022",
  "Operating System": {
    "Platform": 2,
    "ServicePack": "",
    "Version": "10.0.19044.0",
    "VersionString": "Microsoft Windows NT 10.0.19044.0"
  },
  "Environment Variables": {
    "USER": null,
    "TEMP": "C:\\Users\\user\\AppData\\Local\\Temp",
    "TMP": "C:\\Users\\user\\AppData\\Local\\Temp",
    "TMPDIR": null,
    "USERPROFILE": "C:\\Users\\user"
  },
  "HID Devices": [
    {
      "DevicePath": "\\\\?\\hid#vid_09da&pid_9090&mi_00&col02#8&3b84ee0a&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "Manufacturer": "A4TECH",
      "ProductName": "USB Device",
      "SerialNumber": "",
      "FriendlyName": "USB Device",
      "VendorID": 2522,
      "ProductID": 37008,
      "InputReportLength": 2,
      "OutputReportLength": 0,
      "FeatureReportLength": 0,
      "CanOpen": true
    },
    {
      "DevicePath": "\\\\?\\hid#vid_046d&pid_c31c&mi_01&col01#8&1ac9385f&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "Manufacturer": "Logitech",
      "ProductName": "USB Keyboard",
      "SerialNumber": "",
      "FriendlyName": "USB Keyboard",
      "VendorID": 1133,
      "ProductID": 49948,
      "InputReportLength": 2,
      "OutputReportLength": 0,
      "FeatureReportLength": 0,
      "CanOpen": true
[... 1762 lines cut ...]
      "Group": "Display",
      "Message": "DPI Awareness enabled",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2022-10-16T20:55:25.0817043+03:00",
      "Group": "Settings",
      "Message": "The preset directory 'C:\\Users\\user\\AppData\\Local\\OpenTabletDriver\\Presets' has been created",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    },
    {
      "Time": "2022-10-16T20:55:25.0817137+03:00",
      "Group": "Settings",
      "Message": "Presets have been refreshed. Loaded 0 presets.",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    }
  ]
}
```

jamesbt365, 2022-10-16:
Make a pull request switching the report parser to the bamboo report parser.

jamesbt365, 2023-04-07:
Fixed in #2458

PR #2458 (2022-10-17): Change IntuosReportParser to BambooReportParser
- Closes #2457
closed 2023-04-07 by jamesbt365