OpenTabletDriver/OpenTabletDriver #2459: Daemon is unable to start and exits with status 64 but environment variable WAYLAND_DISPLAY and DISPLAY are set.

ssex-dev, 2022-10-19:
# Problem
Daemon is unable to start and exits with status 64.

# Steps
Installed opentabletdriver via AUR and
and run the command from the documentation.

```sh
# Reload the systemd user unit daemon
systemctl --user daemon-reload
# Enable and start the user service
systemctl --user enable opentabletdriver --now
```
# System

OS: Arch Linux x86_64
Kernel: 6.0.2-arch1-1
WM: sway
CPU: AMD Ryzen 9 5900X (24) @ 3.700GHz
GPU: AMD ATI Radeon Pro W5700

# Debug Info

```sh
% systemctl --user status opentabletdriver
× opentabletdriver.service - OpenTabletDriver Daemon
     Loaded: loaded (/usr/lib/systemd/user/opentabletdriver.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Wed 2022-10-19 21:11:04 CEST; 4min 18s ago
   Duration: 1ms
    Process: 3251 ExecStart=/usr/bin/otd (code=exited, status=64)
   Main PID: 3251 (code=exited, status=64)
        CPU: 1ms

Oct 19 21:11:04 Stargate systemd[802]: opentabletdriver.service: Scheduled restart job, restart counter is at 5.
Oct 19 21:11:04 Stargate systemd[802]: Stopped OpenTabletDriver Daemon.
Oct 19 21:11:04 Stargate systemd[802]: opentabletdriver.service: Start request repeated too quickly.
Oct 19 21:11:04 Stargate systemd[802]: opentabletdriver.service: Failed with result 'exit-code'.
Oct 19 21:11:04 Stargate systemd[802]: Failed to start OpenTabletDriver Daemon.
```
```sh
% env | grep -i display
WAYLAND_DISPLAY=wayland-1
DISPLAY=:0
```
```sh
% ls $XDG_RUNTIME_DIR/wayland-*                                                                                    3 ↵
/run/user/1000/wayland-1  /run/user/1000/wayland-1.lock
```
```sh
% ls /tmp/.X11-unix/
X0
```

Any idea what I could try?

jamesbt365, 2022-10-19:
Check systemctl --user show-environment

This will show the envvars that the user session has.

<sup>also use the templates in the future thanks</sup>

ssex-dev, 2022-10-19:
Thank you very much for the fast reply. It is not listed in there,  I didn't know that envvars would be different in a user session.
Sorry about the template.

jamesbt365, 2022-10-19:
systemctl --user import-environment WAYLAND_DISPLAY

jamesbt365, 2022-10-19:
You can probably just throw this in your wm startup if it keeps happening, may also need to put DISPLAY on it too.

closed 2022-10-19 by jamesbt365

ssex-dev, 2022-10-19:
I got it working with 
```sh
systemctl --user import-environment WAYLAND_DISPLAY
```
Thanks again!