OpenTabletDriver/OpenTabletDriver #2468: Support Request: OpenTabletDriver won't open post-installation due to missing (presumably) dotnet library.

Cr-Un, 2022-10-24:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### What do you need help with?

I installed OTD after a fresh Pop!_OS 22.04 LTS installation and it wouldn't open. Running "opentabletdriver" in terminal gave this output:
```
A fatal error occurred. The required library libhostfxr.so could not be found.
If this is a self-contained application, that library should exist in [/usr/share/OpenTabletDriver/].
If this is a framework-dependent application, install the runtime in the global location [/usr/share/dotnet] or use the DOTNET_ROOT environment variable to specify the runtime location or register the runtime location in [/etc/dotnet].

The .NET runtime can be found at:
  - https://aka.ms/dotnet-core-applaunch?missing_runtime=true&arch=x64&rid=pop.22.04-x64&apphost_version=6.0.4
```

How should I go about installing it properly?

### What have you tried?

I've reinstalled the OS for a clean plate, followed the installation guide for Debian/Ubuntu, downloaded the appropriate Microsoft repository package and did every steps one by one. I also tried setting environment variables via `.bashrc` without success.
A couple years ago someone opened an issue similar to this but it was closed without further responses, I wonder if this is a niche problem.

### Diagnostic Information

N/A

jamesbt365, 2022-10-24:
This seems to be a problem with distros that use apt and the latest dotnet version supplied by that.

Downgrade using this:
`sudo apt install aspnetcore-runtime-6.0=6.0.8-1 dotnet-apphost-pack-6.0=6.0.8-1 dotnet-host=6.0.8-1 dotnet-hostfxr-6.0=6.0.8-1 dotnet-runtime-6.0=6.0.8-1 dotnet-sdk-6.0=6.0.400-1 dotnet-targeting-pack-6.0=6.0.8-1`

daishishi, 2023-02-04:
> This seems to be a problem with distros that use apt and the latest dotnet version supplied by that.
>
> Downgrade using this: `sudo apt install aspnetcore-runtime-6.0=6.0.8-1 [...]

This doesn't work on Pop! OS 22.04 LTS. What now?
```E: Versão '6.0.8-1' para 'aspnetcore-runtime-6.0' não foi encontrada
E: Versão '6.0.8-1' para 'dotnet-apphost-pack-6.0' não foi encontrada
E: Versão '6.0.8-1' para 'dotnet-host' não foi encontrada
E: Versão '6.0.8-1' para 'dotnet-hostfxr-6.0' não foi encontrada
E: Versão '6.0.8-1' para 'dotnet-runtime-6.0' não foi encontrada
E: Versão '6.0.400-1' para 'dotnet-sdk-6.0' não foi encontrada
E: Versão '6.0.8-1' para 'dotnet-targeting-pack-6.0' não foi encontrada
```

jamesbt365, 2023-02-04:
https://learn.microsoft.com/en-us/dotnet/core/install/linux-package-mixup#solutions