OpenTabletDriver/OpenTabletDriver #2472: Calling Driver.Detect results in junk input data being reported

gutyina70, 2022-10-31:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Steps to Reproduce

Call Driver.Detect and subscribe to all endpoint's Report event (and make sure the tablet doesn't send reports by moving the stylus on it etc):
```
using Microsoft.Extensions.DependencyInjection;
using OpenTabletDriver;
using OpenTabletDriver.Desktop;
using OpenTabletDriver.Desktop.Interop.AppInfo;
using OpenTabletDriver.Desktop.Reflection;

var driver = new DesktopServiceCollection()
  .AddSingleton<Driver>()
  .AddSingleton<IPluginManager, PluginManager>()
  .AddSingleton<IAppInfo, AppInfo>()
  .BuildServiceProvider()
  .GetRequiredService<Driver>();

driver.Detect();

// Thread.Sleep(50); // this fixes the issue, but there must be a better way right?

foreach(var device in driver.InputDevices)
{
  foreach(var endpoint in device.Endpoints)
  {
    endpoint.Report += (sender, e) => Console.WriteLine("Got report");
  }
}

Console.ReadKey();
```

### Expected Behavior

No reports should've been sent.

### Observed Behavior

Got 1 report, because Detect asyncronously [sends a report](https://github.com/OpenTabletDriver/OpenTabletDriver/blob/master/OpenTabletDriver/InputDeviceEndpoint.cs#L82) to test the tablet I assume.

### Tablet Device

VEIKK S640

### Diagnostic Information

[otd-diagnostics.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/9905480/otd-diagnostics.txt)

gutyina70, 2022-11-04:
The first report I get with the above code has these raw bytes: `129 4 0 0 0 0 0 0`.

X9VoiD, 2022-11-04:
There isn't enough documentation about Veikk tablets' reports so also try to get the report format for the following kinds of input:

- Pen input without pen tip pressed
- Pen input with pen tip pressed
- Pen buttons without pen tip pressed
- Going in and out of range

Unfortunately your tablet do not have buttons on it so we'll make do with what we can grab.

gutyina70, 2022-11-04:
The tablet debugger shows this for the same report:
[image]

X9VoiD, 2022-11-04:
> try to get the report format for the following kinds of input

Specifically, I'd like to know the first and second byte for each kind of report.

gutyina70, 2022-11-04:
- Pen input without pen tip pressed: `41 A0`
 - Pen input with pen tip pressed `41 A1`
 - Pen button 1 without pen tip pressed: `41 A2`
 - Pen button 2 without pen tip pressed: `41 A4`
 - Going in range: `41 A0`
 - Going out of range: `41 C0`

gonX, 2022-11-21:
The PR is only for backporting to 0.6.x - we still need it on `master`. See #2493.

X9VoiD, 2022-12-31:
Fixed by #2555

PR #2475 (2022-11-17): Fix incorrectly parsed ack report in Veikk S640
Other Veikk tablets that use SkipByteReportParser should be tested if they also use the same format as this.

Fixes #2472 

###### should also cherry-pick this to master at some point in the future.
closed 2022-12-31 by X9VoiD