OpenTabletDriver/OpenTabletDriver #2473: Tablet detected however cursor does not move on MacOS Ventura 13.0

quantified-null, 2022-11-01:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Steps to Reproduce

1. Download and install OpenTabletDriver into Applications folder on a Mac running MacOS Ventura 16.0.
2. Launch OpenTabletDriver.app
3. Wait for tablet to be detected
4. Move the pen on the tablet
5. No cursor movement is produced

### Expected Behavior

The tablet is detected, and when the pen is moved, the cursor will move.

### Observed Behavior

The cursor did not move, however osu!lazer (which I believe uses OpenTabletDrivers) does move the cursor in game and the tablet debugger works as intended. (See below)

[image]

### Tablet Device

Gaomon S620

### Diagnostic Information

{
  "App Version": "OpenTabletDriver v0.6.0.4",
  "Build Date": "05/04/2022",
  "Operating System": {
    "Platform": 4,
    "ServicePack": "",
    "Version": "13.0.0",
    "VersionString": "Unix 13.0.0"
  },
  "Environment Variables": {
    "USER": "quantified_null",
    "TEMP": null,
    "TMP": null,
    "TMPDIR": "/var/folders/ng/y2c7wmkj1ys6ln9vck72tms40000gn/T/"
  },
  "HID Devices": [
    {
      "DevicePath": "IOService:/AppleARMPE/arm-io@10F00000/AppleT810xIO/usb-drd0@82280000/AppleT8103USBXHCI@00000000/usb-drd0-port-hs@00100000/IOUSBHostDevice@00100000/AppleUSB20Hub@00100000/AppleUSB20HubPort@00120000/USB2.0 Hub@00120000/AppleUSB20Hub@00120000/AppleUSB20HubPort@00122000/SteelSeries Arctis 5@00122000/IOUSBHostInterface@5/AppleUserUSBHostHIDDevice",
      "Manufacturer": "SteelSeries",
      "ProductName": "SteelSeries Arctis 5",
      "SerialNumber": "00000000",
      "FriendlyName": "SteelSeries Arctis 5",
      "VendorID": 4152,
      "ProductID": 4778,
      "InputReportLength": 35,
      "OutputReportLength": 39,
      "FeatureReportLength": 0,
      "CanOpen": true
    },
    {
      "DevicePath": "IOService:/AppleARMPE/arm-io@10F00000/AppleT810xIO/usb-drd0@82280000/AppleT8103USBXHCI@00000000/usb-drd0-port-hs@00100000/IOUSBHostDevice@00100000/AppleUSB20Hub@00100000/AppleUSB20HubPort@00150000/IOUSBHostDevice@00150000/IOUSBHostInterface@0/AppleUserUSBHostHIDDevice",
      "Manufacturer": "Unknown Manufacturer",
      "ProductName": "Unknown Product Name",
      "SerialNumber": "",
      "FriendlyName": "Unknown Product Name",
      "VendorID": 1105,
      "ProductID": 33535,
      "InputReportLength": 1028,
      "OutputReportLength": 1029,
[... 1777 lines cut ...]
      "Time": "2022-11-01T17:10:09.543196+13:00",
      "Group": "DeviceIOException",
      "Message": "No serial number.",
      "StackTrace": "   at HidSharp.Platform.MacOS.MacHidDevice.GetSerialNumber()\n   at OpenTabletDriver.Devices.HidSharpBackend.HidSharpEndpoint.<>c.<get_SerialNumber>b__19_0(HidDevice d)\n   at OpenTabletDriver.Extensions.TryGet[TSource,TValue](TSource source, Func`2 predicate, TValue& value)",
      "Level": 3,
      "Notification": false
    },
    {
      "Time": "2022-11-01T17:10:09.543204+13:00",
      "Group": "DeviceIOException",
      "Message": "No serial number.",
      "StackTrace": "   at HidSharp.Platform.MacOS.MacHidDevice.GetSerialNumber()\n   at OpenTabletDriver.Devices.HidSharpBackend.HidSharpEndpoint.<>c.<get_SerialNumber>b__19_0(HidDevice d)\n   at OpenTabletDriver.Extensions.TryGet[TSource,TValue](TSource source, Func`2 predicate, TValue& value)",
      "Level": 3,
      "Notification": false
    },
    {
      "Time": "2022-11-01T17:10:09.543213+13:00",
      "Group": "DeviceIOException",
      "Message": "No serial number.",
      "StackTrace": "   at HidSharp.Platform.MacOS.MacHidDevice.GetSerialNumber()\n   at OpenTabletDriver.Devices.HidSharpBackend.HidSharpEndpoint.<>c.<get_SerialNumber>b__19_0(HidDevice d)\n   at OpenTabletDriver.Extensions.TryGet[TSource,TValue](TSource source, Func`2 predicate, TValue& value)",
      "Level": 3,
      "Notification": false
    }
  ]
}

jamesbt365, 2022-11-01:
1. Check is otd is checked in input monitoring & accessibility in privacy and control, if it is in there, remove it by click "-" button, then add it back.
2. Open activity monitor and kill Opentabletdriver/opentabletdriver.daemon
3. Make sure otd is moved to applications folder

quantified-null, 2022-11-03:
"1." Fixed the issue. 

Side note: The FAQ should probably say this rather than uncheck and recheck the options as turning it off and on again does not fix it. Thanks.

closed 2022-11-03 by quantified-null