OpenTabletDriver/OpenTabletDriver #2481: Delay on pen button inputs with HawkuFilters applied

lenforiee, 2022-11-06:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Steps to Reproduce

Set variables on hawku filter to 15ms and 144hz (could happen on other settings too)

### Expected Behavior

The movement input should be smoothed however pen button inputs should happen without any delays applied and blocks (the button stay clicked for few seconds and blocks other button inputs)

### Observed Behavior

The movement is indeed smoothed but pen button inputs are happening with a delay and with the blocks. If I change the values to 10ms/144hz the problem is minimalistic but blocks still happens.

PS: I am opening the issue on the driver repo and not the filter itself because I have read the filter code and I do not see anything wrong (unless I am missing something)

### Tablet Device

Wacom CTL-480

### Diagnostic Information

[diagnostic.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/9946972/diagnostic.txt)

X9VoiD, 2022-11-06:
Are you setting frequency to 144hz? That's incredibly low. Unless you're on a potato pc you should use 1000hz.

lenforiee, 2022-11-06:
These settings have been a recommendation from a friend, with the premise of emulating the feel of wacom drivers. And although they do the job, if the settings are problem I am open on feedback, however I feel like in reality this should not complicate with how pen buttons work.

jamesbt365, 2022-11-06:
Those settings are forcing the fallback timer, just use 250, 500 or 1000.

closed 2022-11-07 by gonX