OpenTabletDriver/OpenTabletDriver #2484: GUI entirely stops trying to connect to daemon after timeout

gonX, 2022-11-07:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Steps to Reproduce

- Start OTD daemon
- Start OTD GUI
- Stop daemon
- Wait for GUI to complain about being unable to connect to the daemon
- Start daemon
- GUI is now softlocked

### Expected Behavior

User can press on GUI's new "Retry connection" button

### Observed Behavior

User must restart GUI

### Tablet Device

N/A

### Diagnostic Information

N/A

jamesbt365, 2024-07-30:
The GUI should automatically close currently right now, it shouldn't deadlock unless the behaviour has changed on recent master?

gonX, 2025-01-24:
Fixed on `avalonia` branch

closed 2025-01-24 by gonX