OpenTabletDriver/OpenTabletDriver #2490: OTD giving me an ms store redirect that doesnt work upon opening

Brainlit04, 2022-11-11:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### What do you need help with?

The tablet driver doesn't seem to open properly as per the instructions provided and shows my an ms store redirect that doesnt go anywhere images provided below, however I am using windows 11 and cannot return to windows 10.

[image]

### What have you tried?

I have tried following the redirect, uninstalling and reinstalling as per the instructions provided and I have also unzipped the files as told, however I found that for a while before the most recent update to windows 11 as of me writing this, If i did not unzip the 2 files provided the driver would work just fine however that is no longer the case.

### Diagnostic Information

Cannot launch app, cannot provide diagnostics.

jamesbt365, 2022-11-11:
Try extracting the archive with a good program like 7zip

Brainlit04, 2022-11-11:
> Try extracting the archive with a good program like 7zip

This worked, thank you very much.

closed 2022-11-11 by jamesbt365