OpenTabletDriver/OpenTabletDriver #2491: Huion Tablet H430P is recognized, but nothing actually works on Linux Mint Vanessa

Semanual, 2022-11-13:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Steps to Reproduce

1. Install OTD from [the official website](https://opentabletdriver.net/Wiki/Install/Linux) with the Ubuntu's guide;
2. Install the downgraded version of dotnet (as mentioned in #2468);
3. Execute `opentabletdriver` on terminal;
4. Pen works normally;
5. Open osu!lazer and start a game;
6. Neither the pen nor the tablet buttons work anymore on the entire OS.

### Expected Behavior

The pen and the tablet's buttons should continue working normally in-game and out of it.

### Observed Behavior

The pen stopped working, the tablet's buttons also have no effect (even after configuring them on Auxiliary Settings), the console outputs no error, and the driver correctly recognizes my tablet's model.

### Tablet Device

Huion H430P

### Diagnostic Information

[diagnostics.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/9996286/diagnostics.txt)

jamesbt365, 2022-11-13:
Reboot. And next time actually look for duplicates...

Semanual, 2022-11-13:
Sorry, I forgot to mention. I already rebooted, twice actually.

Semanual, 2022-11-13:
@jamesbt365 can you please reopen it, as it wasn't actually solved? And yes, I tried rebooting again and the same problem still happens.

InfinityGhost, 2022-11-13:
> Reboot. And next time actually look for duplicates...

Mention the issue it is a duplicate of when you close issues for this.

gonX, 2022-11-13:
Reopening and converting to support request for now. There is an ongoing Discord conversation [here](https://discord.com/channels/615607687467761684/723399565780189234/1041161264254177340)

gonX, 2023-11-19:
This is an old support request, so I will be closing it as stale as version 0.6.3.0 of OpenTabletDriver has released since then, and OP uses 0.6.0.4. Please try with the newest release version, and if it still doesn't work, please feel free to reopen the issue.

closed 2023-11-19 by gonX