OpenTabletDriver/OpenTabletDriver #2497: Driver has inconsistent detection (Huion Kamvas 13)

joshuah345, 2022-11-14:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Steps to Reproduce

1. Launch OpenTabletDriver with tablet connected
2. Try to detect tablets again

### Expected Behavior

Tablet gets detected with no errors.

### Observed Behavior

On the first launch of OTD, the tablet gets detected with no errors, but using cmd/crtl+d to detect again results in a combination of these errors:

> [DeviceIOException:Error]	Failed to open USB device: 0xE00002C5
   at HidSharp.Platform.MacOS.MacHidDevice.GetDeviceString(Int32 index)
   at OpenTabletDriver.Driver.DeviceMatchesStrings(IDeviceEndpoint device, IDictionary`2 deviceStrings)

>[Exception:Error]	Failed retrieving USB Device
   at HidSharp.Platform.MacOS.MacHidDevice.GetDeviceString(Int32 index)
   at OpenTabletDriver.Driver.DeviceMatchesStrings(IDeviceEndpoint device, IDictionary`2 deviceStrings)

> [Exception:Error]	Plugin Interface creation failed. 0xE00002BE
   at HidSharp.Platform.MacOS.MacHidDevice.GetDeviceString(Int32 index)
   at OpenTabletDriver.Driver.DeviceMatchesStrings(IDeviceEndpoint device, IDictionary`2 deviceStrings)

after these errors occur, the tablet does not get detected again until otd is relaunched.

### Tablet Device

Huion Kamvas 13 

### Diagnostic Information

[Untitled.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/10006677/Untitled.txt)

jamesbt365, 2022-11-14:
Can you get a few string dumps replugging the tablet in between?

tablet > device string reader
9580
109
empty
dump all

joshuah345, 2022-11-14:
> Can you get a few string dumps replugging the tablet in between?
> 
> tablet > device string reader 9580 109 empty dump all

This doesn't seem to be working. even after replugging the tablet it asks me to do it again

jamesbt365, 2023-08-23:
Could you try on windows?

jamesbt365, 2025-01-24:
Stale

gonX, 2026-03-13:
FWIW, I have a feeling that OpenTabletDriver is the one grabbing exclusive access to the device.