OpenTabletDriver/OpenTabletDriver #2498: Support Request: cant adjust tablet area (xp-pen g640)

stargazer225, 2022-11-15:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### What do you need help with?

i cant adjust my tablet area so now i have to play osu with full area and high ingame sensitivity which adds input lag and makes my aim very shaky in comparison to when i use a small area with opentabletdrivers.

### What have you tried?

reinstalling all drivers/vmulti, changing monitors etc, i basically tried everything common

### Diagnostic Information

N/A

jamesbt365, 2022-11-15:
Diagnostic information is applicable here.

stargazer225, 2022-11-16:
no i cant upload a .json file and im not experienced enough to reformat it or whatever u do with it

jamesbt365, 2022-11-16:
it says to change the file extension to .txt or paste it in the box.

N/A means non applicable, not that you don't know how.

stargazer225, 2022-11-16:
there, its .txt now

stargazer225, 2022-11-16:
im new to github, is there any news or tips for me what i can try to do to get it working? ive been struggling with this problem for weeks now and this is my last resort...

jamesbt365, 2022-12-02:
- Open "Device Manager"
- At the top click view > device by container
- Look at the list of devices
- Find your tablet
- Open the dropdown menu (container) next to the tablet
- In the drop down container, find any of the following and right click them: <OEM> tablet (e.g. “Wacom Tablet”; “Gaomon Tablet”), the model number (e.g. “CTL-472”), or the name of the tablet (“Wacom Intuos S”)
- Click on update -> browse my computer for drivers -> let me pick from a list of available drivers -> usb input device, usb composite device, or hid compliant device -> next
- Repeat the first two steps if any other matching devices are present
- Replug your tablet and check if otd detects it

jamesbt365, 2023-04-07:
Closing as stale, the proper solution would be to uninstall the drivers and cleanup the remnants. You can use [TabletDriverCleanup](https://github.com/X9VoiD/TabletDriverCleanup/releases/tag/v4.0.2) for this.

closed 2023-04-07 by jamesbt365