OpenTabletDriver/OpenTabletDriver #2501: Cannot bind pen buttons Huion HC16

IgnisFlamme, 2022-11-18:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Steps to Reproduce

1. Try to bind the two pen buttons on the stylus
2. Fail

### Expected Behavior

I should be able to bind the two pen buttons

### Observed Behavior

No way to bind the pen buttons despite the two buttons being detected.

### Tablet Device

Huion HC16

### Diagnostic Information

[diagnostics.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/10045009/diagnostics.txt)

IgnisFlamme, 2022-11-18:
I checked in the tablet diagnostics and it detects the two pen buttons, so now I'm confused why I cant rebind it to a keyboard shortcut.

jamesbt365, 2023-08-23:
Please update OTD and your system and check if this still doesn't work.

gonX, 2023-10-16:
This looks like it should be working in 0.6.3.0 - the pen has 2 buttons associated in its configuration.

Please reopen this issue if the issue still occurs with 0.6.3.0.

closed 2023-10-16 by gonX