OpenTabletDriver/OpenTabletDriver #2513: Support Request: Gaomon S56k detected,but only with windows ink

T33XX, 2022-11-27:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### What do you need help with?

So, as I said in the title, my gaomon s56k only gets detected by ost with interface 2 but i think (i'm not sure) windows ink takes over, I can't use my cursor properly, it's very slow and it's shaped like a dot with a crosshair and in osu it just gets stuck at the top left corner.

### What have you tried?

I tried anything I could find on discord, switching between interfaces, restarting replugging, cleaning up old drivers junk,and probably unecessary stuff like reinstalling the drivers, manually updating the driver with usb device

### Diagnostic Information

{
  "App Version": "OpenTabletDriver v0.6.0.4",
  "Build Date": "5/4/2022",
  "Operating System": {
    "Platform": 2,
    "ServicePack": "",
    "Version": "10.0.19043.0",
    "VersionString": "Microsoft Windows NT 10.0.19043.0"
  },
  "Environment Variables": {
    "USER": null,
    "TEMP": "C:\\Users\\ADMINI~1\\AppData\\Local\\Temp",
    "TMP": "C:\\Users\\ADMINI~1\\AppData\\Local\\Temp",
    "TMPDIR": null,
    "USERPROFILE": "C:\\Users\\Administrator"
  },
  "HID Devices": [
    {
      "DevicePath": "\\\\?\\usb#vid_256c&pid_006e&mi_02#7&29a3de77&0&0002#{dee824ef-729b-4a0e-9c14-b7117d33a817}",
      "Manufacturer": null,
      "ProductName": "Tablet Monitor",
      "SerialNumber": "Unknown Serial Number",
      "FriendlyName": "Tablet Monitor",
      "VendorID": 9580,
      "ProductID": 110,
      "InputReportLength": 8,
      "OutputReportLength": 0,
      "FeatureReportLength": 0,
      "CanOpen": true
    },
    {
      "DevicePath": "\\\\?\\hid#vid_046d&pid_c084&mi_01&col01#8&39142b5c&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\\kbd",
      "Manufacturer": "Logitech",
      "ProductName": "G102 Prodigy Gaming Mouse",
      "SerialNumber": "097638733532",
      "FriendlyName": "G102 Prodigy Gaming Mouse",
      "VendorID": 1133,
      "ProductID": 49284,
      "InputReportLength": 9,
      "OutputReportLength": 0,
      "FeatureReportLength": 0,
      "CanOpen": false
    },
    {
      "DevicePath": "\\\\?\\hid#vid_256c&pid_006e&mi_00&col01#8&1b97c612&1&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "Manufacturer": null,
      "ProductName": "Tablet Monitor",
      "SerialNumber": "",
      "FriendlyName": "Tablet Monitor",
      "VendorID": 9580,
      "ProductID": 110,
      "InputReportLength": 8,
      "OutputReportLength": 0,
[... 1907 lines cut ...]
      "Time": "2022-11-27T13:39:40.849155+02:00",
      "Group": "IPC",
      "Message": "Connected to a client.",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2022-11-27T13:39:40.849167+02:00",
      "Group": "Display",
      "Message": "DPI Awareness enabled",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2022-11-27T13:39:40.8491788+02:00",
      "Group": "Settings",
      "Message": "Presets have been refreshed. Loaded 0 presets.",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    }
  ]
}

jamesbt365, 2022-11-27:
This is going to be one of the annoying variants. It will be significantly faster if you join our [Discord](https://opentabletdriver.net/Discord)

gonX, 2023-11-19:
Closing issue as stale due to age. It's worth noting you are using the Administrator account, which is not officially supported.

closed 2023-11-19 by gonX