OpenTabletDriver/OpenTabletDriver #2518: Support Request: Cant change area/ wouldnt use it

Dauerst0ned, 2022-12-02:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### What do you need help with?

After being gone for a month i want to play osu but my are isnt working.. i tried to change it. Also i reinstalled it and doenst work either. How can i fix it, what is the problem

### What have you tried?

Reinstalling, restart pc.

### Diagnostic Information

[diagnostic.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/10141926/diagnostic.txt)

jamesbt365, 2022-12-02:
- Open "Device Manager"
- At the top click view > device by container
- Look at the list of devices
- Find your tablet
- Open the dropdown menu (container) next to the tablet
- In the drop down container, find any of the following and right click them: <OEM> tablet (e.g. “Wacom Tablet”; “Gaomon Tablet”), the model number (e.g. “CTL-472”), or the name of the tablet (“Wacom Intuos S”)
- Click on update -> browse my computer for drivers -> let me pick from a list of available drivers -> usb input device, usb composite device, or hid compliant device -> next
- Repeat the first two steps if any other matching devices are present
- Replug your tablet and check if otd detects it

Also there is considerable amount of duplicates

closed 2022-12-02 by jamesbt365

Dauerst0ned, 2022-12-02:
where is device manager i cant find it

jamesbt365, 2022-12-02:
an application on your computer.

Dauerst0ned, 2022-12-02:
bruh

Dauerst0ned, 2022-12-02:
maybe its also another problem because its still doesnt work
The thing is i have 2  monitors one is 24 and one 27 and if i use display 1 as area it still goes to display 2 a little bit

jamesbt365, 2022-12-02:
Screenshot device manager with the container for your tablet open

Dauerst0ned, 2022-12-02:
[image]
im Germany so sry if you cant understand something

Dauerst0ned, 2022-12-02:
there was also this error
[image]

Dauerst0ned, 2022-12-02:
hey i checked the problem with my other tablet. and the problem is definitly not my tablet ... / driver

jamesbt365, 2022-12-02:
right click hid complaint mouse > update > browse my computer > let me pick from a list of available drivers > screenshot.

Dauerst0ned, 2022-12-02:
[image]

jamesbt365, 2022-12-02:
bottom one -> next. Once it is done replug your tablet.

Dauerst0ned, 2022-12-02:
still doesnt work.. the thing is technically i have right area/ sens but it doesnt stop there where i limited it and it is postponed

Dauerst0ned, 2022-12-02:
like if i go in the middle it is on right top

Dauerst0ned, 2022-12-03:
so i tried it on another pc and it also doesnt work. So it should be problems with the tablet drivers. Right?

jamesbt365, 2022-12-03:
No. Windows has been pushing broken device drivers for months now.

jamesbt365, 2022-12-03:
Screenshot device manager again.

Dauerst0ned, 2022-12-03:
[image]

Dauerst0ned, 2022-12-03:
[image]

Dauerst0ned, 2022-12-03:
i first it maybe was win 11 
but on win 10  also doesnt work

jamesbt365, 2022-12-03:
right click ctl-472 > update > browse my computer > let me pick from a list of available drivers > usb input/composite device > next > finish > replug.

Dauerst0ned, 2022-12-03:
wanst the problem

Dauerst0ned, 2022-12-03:
[image]
u see my are and if i go to the corner it goes on my second monitor

jamesbt365, 2022-12-03:
Join the [Discord](https://discord.com/invite/9bcMaPkVAR) server. I'll deal with this there.

Dauerst0ned, 2022-12-03:
i joined