OpenTabletDriver/OpenTabletDriver #2519: No tablets were detected with warning "Another tablet driver found: Gaomon"

nanakura, 2022-12-04:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Steps to Reproduce

1. Uninstall original Gaomon tablet driver
2. Restart windows
3. Open OpenTabletDriver.UX.Wpf.exe

### Expected Behavior

detect Gaomon tablet 

### Observed Behavior

no tablets detected and display some warning
[image]
But I have uninstalled the official Gaomon tablet driver

### Tablet Device

Gaomon M6

### Diagnostic Information

[diagnostics.json.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/10147979/diagnostics.json.txt)

jamesbt365, 2022-12-04:
There is no configuration for this tablet.

tablet > device string reader
9580
109
dump all, save that and send here.

nanakura, 2022-12-04:
> There is no configuration for this tablet.
> 
> tablet > device string reader 9580 109 dump all, save that and send here.

[dumps.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/10148282/dumps.txt)
It looks like it has an encoding problem...

jamesbt365, 2022-12-04:
thats just how it looks.

jamesbt365, 2022-12-04:
- Make a `Configurations` folder next to the executables of otd
- Download [this](https://cdn.discordapp.com/attachments/628651971267788800/1048913443614380042/M6.json) file
- Move it into that folder.

jamesbt365, 2022-12-04:
Once you have done that, it should detect but we aren't done yet.

nanakura, 2022-12-04:
It worked

jamesbt365, 2022-12-04:
we aren't done.

jamesbt365, 2022-12-04:
- Go into otd
- Go to `tablet > tablet debugger`
- Move your pen to the bottom right of your tablet (ignore the on screen visualiser)
- Get the maximum x and y coordinates that you can and the max pressure.
- Check if the pen buttons go to true when pressing them
- Bind a few of the tablet buttons and check if they work.

nanakura, 2022-12-04:
[image]
tablet buttons are working

closed 2022-12-04 by nanakura